sobri909/LocoKit #1: Getting Compile Errors: Generic parameter 'ML' is not used in function signature

dmr07, 2017-08-13:
Looking to try it out but running into some problems on the demo app. 

I ran `pod install` before building. Running into some compile problems:

[image]

sobri909, 2017-08-14:
Ah, yeah, the stable branch o [Upsurge](https://github.com/aleph7/Upsurge) isn't ready for Xcode 9 yet. Though there is [a Swift 4 branch](https://github.com/aleph7/Upsurge/tree/xcode9) that's ready to go. 

I should actually remove the Upsurge dependency from the demo app. It's just there because I wanted to get things built fast, so it let me reuse some existing accelerated code. But I think there's only a small handful of lines in the demo app that're making use of Upsurge, so it won't be a major task to remove the dependency. 

I'll have the next major ArcKit release ready to go this week hopefully (with the machine learning features). So when I ship that I'll make sure the demo app is ready to go on Xcode 9. 

Though if you want to get it going now in Xcode 9, you might possibly be able to just comment out of handful of lines that're using Upsurge.

dmr07, 2017-08-14:
Makes sense. 

In that case, I'll wait until the next release.. I'm more interested in seeing an implementation of the Kalman filter. I doing a school project and kinda stuck. 

I took a quick glance at the code, and I'm mostly seeing an average. Is the Kalman filter implementation on another branch?

sobri909, 2017-08-14:
ArcKit's Kalman filter is in the framework binary. As you've probably discovered, they're not easy to write! So I've kept that code proprietary.

If your school project is for writing a Kalman filter, then my advice would be to aim for the absolute simplest Kalman possible, eliminating all inputs except the absolute minimum required. Get that bare basic algorithm working, then add more to the matrices from there.

sobri909, 2017-09-15:
I've just pushed ArcKit v2.0.0, built on Swift 4 in Xcode 9 GM, so this issue is now resolved.

closed 2017-09-15 by sobri909