sobri909/LocoKit #2: License

Xioshock, 2017-08-16:
Hi, do you have any public license for this pod. Something like MIT?

sobri909, 2017-08-17:
The SDK is under a commercial license, although all currently released features are (and will continue to be) free for unlimited use. 

The next round of features (machine learning based activity type detection) will require a service subscription. Although there will most likely also be a free tier for the machine learning models too, allowing for use on a limited number of devices, or something like that. I'll be working out all those details over the next couple of months. 

Anyway, the tl;dr is that all the existing location / motion recording features are free to use in any project without restriction.

sobri909, 2017-08-17:
Oh, and apologies for the lack of clear explanation of this in the current readme! I'm still only a team of one, so I've been focusing my limited time on getting the functionality built and documented first.

Xioshock, 2017-08-17:
I see. Thanks you.

closed 2017-08-17 by Xioshock