sobri909/LocoKit #3: CoreMotion Question

felix-dumit, 2017-09-13:
Is the CoreMotion data used to improve the location accuracy or just used to provide filtered and sanitised accelerometer, etc ?

sobri909, 2017-09-13:
The Core Motion data is used to fill in the extra properties in the [LocomotionSamples](https://sobri909.github.io/ArcKit/Classes/LocomotionSample.html#/Motion%20Properties), and is also fed into the machine learning classifier (which I'll be publishing a beta of tomorrow-ish). 

I don't use any of the Core Motion data to help correct location data. The location data filtering / smoothing is done by a couple of Kalman filters, as a first pass, then a second pass is done by a dynamic smoothing / averaging algorithm.

sobri909, 2017-09-15:
To answer this question a bit better, a few hours ago I pushed ArcKit v2 live, which makes use of the Core Motion data (and Core Location data) to do machine learning activity type classifying :) 

https://github.com/sobri909/ArcKit/wiki/Introduction-to-ArcKit-v2.0.0

I've written up a wiki article giving a brief introduction. But I'm still working on writing up the new API docs and some more comprehensive howtos and tutorials and the like. So there's more coming soon...

closed 2017-10-03 by sobri909