sobri909/LocoKit #4: Accuracy question

freeubi, 2017-09-20:
Hello,

i'm planning to use this framework in an exercise tracking application. 
I have tried draw the exercise track with the filteredPoints and the locomotionSample.location

The locomotionSample definitely smoother, but i got a strange wave - i!m walking on the streets but the line is not correct - see the attached image.

Can you give any advice to get more accurate results? 
I'm useing an iphone 7 plus with mobile data connection.
[image]

sobri909, 2017-09-20:
Hey :) 

The first thing I'd do is to determine what the best available location accuracy in the area is.

For the area in your screenshot it looks like those are city streets with buildings on either side, which means the best possible location accuracy will be between 10 and 65 metres (from a combination of weak GPS and wifi triangulation), perhaps averaging around 30 metres. 

But to see it in real time I'd run the ArcKit Demo App while walking around the area and see what sort of numbers you get for "Receiving accuracy". 

If that area isn't averaging much better than about 30 metres accuracy, there's not much hope unfortunately. Your options are either incorrect paths that jump around erratically (ie the raw data or the filtered data), or incorrect paths that weave smoothly around but are still mostly wrong (ie the smoothed samples). The device is unlikely to be able to do better without clear GPS line of sight. 

Though one thing that you'll probably notice is that if you walk the same route several times, the accuracy should improve each time, due to iOS collecting more comprehensive wifi triangulation data each time. 

The second thing I'd look into is the LocomotionManager's `maximumDesiredLocationAccuracy`. The default value for that is 30 metres, but you could experiment with setting it to `kCLLocationAccuracyBest` and see if that makes any difference. It may push the device to try harder in the more difficult situations (at the expense of higher battery consumption).

freeubi, 2017-09-20:
Thanks, i will check it on my way home :)

The strange thing that i see is that the course almost correct, just a few meters in one way. 
There is an another image. The long straight line is when i'm moving undergrounds then come up, wait in a coffee shop 1 hour, them move.

By the way the framework It does an excellent job to eliminate the in building and waiting GPS jumps.

[image]

sobri909, 2017-09-20:
Thanks :)

Yeah, it's super common to see the location data consistently put you on the wrong side of the road. And if you walk the same way regularly, you'll notice that it often happens at the same places along the trip. The buildings create GPS shadows that cause the location data to be wrong in similar ways each time.

Eventually I'd like to add a "snap-to-roads" feature, that could help in these situations. But that will require access to road databases, which might not be free, and will require some careful heuristics so that it only snaps to roads at appropriate times. So adding that feature isn't in the near term plans yet.

sobri909, 2017-10-06:
Updating on this low-accuracy-in-built-up-areas issue:

I've started looking into feeding road data into the filters and smoothing, to increase path accuracy under building shadows, and I'm seeing some initial promising signs. Though it's probably going to be a few months before I can ship an SDK version with it built in. 

In the meantime, I've stumbled onto [Mapbox Map Matching API](https://www.mapbox.com/help/define-map-matching/) (https://www.mapbox.com/api-documentation/#map-matching), which provides a rather brutal (but aesthetically pleasing) snap-to-roads algorithm. 

The Mapbox snap-to-roads is too aggressive for what I want to achieve, in that it will snap walking paths to roads even if the user walked through a park or through a building, so it will presumably sacrifice correctness for the sake of superficially pleasing results. But it's at least available now, so it's a potential option until ArcKit has something smarter built in.

freeubi, 2017-10-06:
Sorry about the delay.
I have tested with a few settings, currently i'm using
`maximumDesiredLocationAccuracy = 1.0`

With that, i get a pretty good route - but the few meter shifting is still on, see the attachment.
I dont think that we can make it better.
[image]

Thanks for the recommendation, but the snap-to-road is not a good feature for exercise tracking :D
Thanks for your help!

closed 2017-10-06 by freeubi