sobri909/LocoKit #5: ArcKit doesn't build in XCode 8.3

gadgetman4u, 2017-10-27:
I did a pod install for ArcKit which was successful, but when I build, I get the error:

“Swift Language Version” (SWIFT_VERSION) is required to be configured correctly for targets which use Swift. Use the [Edit > Convert > To Current Swift Syntax…] menu to choose a Swift version or use the Build Settings editor to configure the build setting directly.

I did this to upgrade to Swift 3.0, but it still fails to build.  I also changed the build target to be iOS 10.3 instead of iOS 11, but still doesn't build.  Is there anyway to fix this?

Thanks.

gadgetman4u, 2017-10-27:
Actually, this is the detailed error that I get:

[BCEROR]None of your accounts are a member of 'U5N7VG8DUG':  An unexpected error occurred. Xcode cannot find a team matching 'U5N7VG8DUG'.
[BCEROR]No profiles for 'com.bigpaua.ArcKit-Demo-App' were found:  Xcode couldn't find a provisioning profile matching 'com.bigpaua.ArcKit-Demo-App'.
“Swift Language Version” (SWIFT_VERSION) is required to be configured correctly for targets which use Swift. Use the [Edit > Convert > To Current Swift Syntax…] menu to choose a Swift version or use the Build Settings editor to configure the build setting directly.
[BCEROR]Code signing is required for product type 'Application' in SDK 'iOS 10.3'

sobri909, 2017-10-28:
Hi @gadgetman4u :)

ArcKit uses Swift 4, and as such requires Xcode 9. 

Hope that helps!

gadgetman4u, 2017-10-30:
Hi Matt,

OK thanks!

Alvin

gadgetman4u, 2017-10-30:
Hi,

Still having problems after installing XCode 9 and building.

I get these errors:

Code Signing Error: No account for team "U5N7VG8DUG". Add a new account in the Accounts preference pane or verify that your accounts have valid credentials.
Code Signing Error: No profiles for 'com.bigpaua.ArcKit-Demo-App' were found:  Xcode couldn't find any iOS App Development provisioning profiles matching 'com.bigpaua.ArcKit-Demo-App'.
Code Signing Error: Code signing is required for product type 'Application' in SDK 'iOS 11.0'

Do I need to create a new account with team "U5N7VG8DUG" and create a new profile?

sobri909, 2017-10-31:
Hey :) Did you get this figured out? 

If not, you need to change the "Team" for the "ArcKit Demo App" target in General -> Signing. Change it to your own Developer Account, and you should be ready to go.

gadgetman4u, 2017-10-31:
Hi Matt,

I figured it out now.

Thanks.

Alvin

closed 2017-10-31 by sobri909