sobri909/LocoKit #7: why notice “Unable to find a pod with name, author, summary, or description matching `ArcKit`” when I pod search ArcKit

ouzs, 2018-02-05:
hello，i clone code About ‘ArcKit’ ，but couldn't run ，tips can‘t find ’ArcKit‘ ，try ’pod install‘ tips ’Unable to find a specification for `ArcKit (= 5.0.0.pre)`‘. how i can fix it, thx谢谢！！

sobri909, 2018-02-05:
Hm. Interesting. It's happening to me too. 

I'll push a new 5.0.0.pre release now, to see if that will fix it. Give me a few mins...

ouzs, 2018-02-05:
OK，i am waiting for u，  and my clone should lastest version

sobri909, 2018-02-05:
Ok, I've just pushed 5.0.0.pre.4, and it's working successfully in the Demo App.

If you want the new persistent timelines (TimelineManager that stores its TimelineItems in an SQLite database) you will need a second pod line:

```
pod 'ArcKit', '~> 5.0.0.pre'
pod 'ArcKit/LocalStore', '~> 5.0.0.pre'
```

The `LocalStore` subspec adds `PersistentTimelineManager`, `PersistentPath`, `PesistentVisit`, etc.

ouzs, 2018-02-05:
I just download demo and pod install,also tips 'Unable to find a specification for `ArcKit (~> 5.0.0.pre.3)',how can i run demo app..thx,谢谢

sobri909, 2018-02-05:
It's working for me here. Make sure you've got the latest commit (I pushed some Demo App changes a few hours ago), and try a `pod update` if `pod install` still fails.

ouzs, 2018-02-05:
OK，thank u very much

sobri909, 2018-02-05:
No worries. Good luck with your project!

closed 2018-02-05 by sobri909