sobri909/LocoKit #8: Generic Parameter 'C' could not be inferred

siverma, 2018-02-06:
let locationT = CLLocationCoordinate2DMake(CLLocationDegrees(34.531), CLLocationDegrees(-122.02358845))
let classifier = ActivityTypeClassifier(coordinate: locationT)

On the second line Xcode is throwing the following compilation error: Generic parameter 'C' could not be inferred

Any idea how to solve this?

sobri909, 2018-02-07:
Hi. Sorry for the slow reply. I've just got a chance to get to this now. 

It looks like you're using the stable ArcKit release, which means you're getting activity type classifiers which need to know their data source / cache type. Which means you need to do something like this: 

```swift
let classifier = ActivityTypeClassifier<ActivityTypesCache>(coordinate: locationT)
```

Although, depending on your project state and requirements, I recommend upgrading to prerelease ArcKit. 

```ruby
pod 'ArcKit', '~> 5.0.0.pre'
pod 'ArcKit/LocalStore', '~> 5.0.0.pre'
```

The generics for data source / cache are no longer necessary in the latest builds. And there's also a convenience [TimelineClassifier](https://github.com/sobri909/ArcKit/blob/master/ArcKit/Base/Timelines/TimelineClassifier.swift) which wraps up the management of fetching and updating classifiers.

sobri909, 2018-02-07:
For some context on the stability state of ArcKit 5.0.0, I'm looking to ship it as final this week. Tomorrow I'll be working on filling out the missing API docs, then it'll be ready to go. The code itself is passing all my tests, so it's just the documentation shortfall holding the release back at this stage.

siverma, 2018-02-07:
Sounds good! Thanks for the prompt reply.

closed 2018-02-07 by siverma