sobri909/LocoKit #10: LocoKit doesn't build on Xcode 9.3

Velociround, 2018-04-03:
I have many "Swift Compiler Error" in my project after installing it with CocoaPods (57).

I can't even build the LocoKit Demo App because it also gives me all of these errors.

This is the error I get when adding it to my project.
- 'collations' is inaccessible due to 'private' protection level
(and many others of the same type)

These are the errors I get trying to build the LocoKit Demo app (20):
- Type 'ComplexArray<T>' does not conform to protocol 'RangeReplaceableCollection'
- Ambiguous use of 'count'
- Type 'ValueArraySlice<Element>' does not conform to protocol 'Collection'

What am I doing wrong?

sobri909, 2018-04-03:
It sounds like your pods cache has got itself confused. 

Those errors look like what would happen if you used outdated dependencies, so I'd try deleting the Pods folder then doing a `pod update` to force CocoaPods to get the latest of everything.

That should fix it.

Velociround, 2018-04-03:
Wow, thank you very much for the quick reply!

I just did, on this order:
- On Xcode, cleaned the project and its build folder
- `pod cache clean --all`
- `rm -r Pods/`
- `pod update`
- `pod install`

Then tried to build it again. It still gives me the same errors (and same amount of errors on each project). I'm puzzled.

I'm running Xcode Version 9.3 (9E145), trying to build for an (actual) iPhone 7 Plus on iOS 10.3

Velociround, 2018-04-03:
I tried changing the build target to "Generic iOS device" and now I'm getting `No such module 'SwiftNotes'` on LocoKit Demo App, but still the same 57 errors on my project.

Changed the build target to "iPhone 6s Plus (simulator)" and I'm back to the 20 build errors on LocoKit Demo App and 57 errors on mine.

Changed the build target to "iPhone X (simulator)" and still the same 20 build errors on LocoKit Demo App and 57 errors on mine.

I tried using `pod try LocoKit` but it says "failed to find project ArcKit Demo App". _(off-topic: also, there are broken links on the documentation still pointing to [ArcKit](https://www.bigpaua.com/arckit/coverage/all) on [ActivityTypeClassifier](https://www.bigpaua.com/locokit/docs_core/Classes/ActivityTypeClassifier.html). I didn't check others)_

Velociround, 2018-04-03:
I re-did the same steps as before, but this time without `pod install` at the end. It worked on the Demo App. I'm currently testing the LocoKit Demo App on my iPhone.

My own project still has the same problems as before, though, and the build errors are not even on your project, all of them are on "GRDB.swift"... but maybe you've had these problems before and might be able to help me?

Velociround, 2018-04-03:
I'm sorry to bother you, [this workaround](https://stackoverflow.com/questions/46101625/using-cocoapods-mixed-swift3-and-swift4-pods) fixed the problem.

```
# Workaround Cocoapods to mix Swift 3.2 and 4
# Manually add to swift4Targets, otherwise assume target uses Swift 3.2
post_install do |installer|
    installer.pods_project.targets.each do |target|
        target.build_configurations.each do |config|
            config.build_settings['SWIFT_VERSION'] = '4'
        end
    end
end
```

You may close the issue, and I hope other people who see this find it helpful. Thanks for your help.

sobri909, 2018-04-03:
Hm. Modifying the Podfile shouldn't be necessary. There should be an easier fix.

Have a look in your main project's **Build Settings** for **Swift Version**. That value should be inherited automatically by each pod, when CocoaPods generates the workspace.

Velociround, 2018-04-05:
Swift Version was 3.0 on my main project's Build Settings. I changed it to 4.1, executed `pod update` again, reopened Xcode, cleaned my build folder and then tried to run my project. It worked.

However, I have never edited this parameter on my project. The project was created with Swift Version 3.0 by default. I created it on February 15th. Shouldn't Xcode have automatically generated my project with Swift 4.0?

Also, according to the linked StackOverflow question, it is possible to mix pods of very different versions using that workaround (which sets different Swift versions for each pod according to their name). I had just quickly edited it to change all pods, because your pod (and its dependencies) is the only Swift dependency I have on my project.

closed 2018-04-29 by sobri909