sobri909/LocoKit #15: There is not LocoKit in Pods on `pod install`

danijelkecman, 2018-05-28:
Cannot build the project because LocoKit Pod does not contain the lib just a Readme.md file.

sobri909, 2018-05-28:
Can you show me your `Podspec` file? 

The pod itself is certainly valid, and is used regularly.

danijelkecman, 2018-05-28:
It used to work until recently, I think before 5.2.0 update. Since 5.2.0 `pod update` results with Pods/LocoKit folder containing Readme.md file only.

danijelkecman, 2018-05-28:
In both,  *master* and *develop* branches.

sobri909, 2018-05-28:
The develop branch isn't in CocoaPods, and I wouldn't recommend trying to pull a pod from there.

sobri909, 2018-05-28:
Yeah, try deleting the Pods folder and installing fresh. Sounds like a broken pods install somehow.

sobri909, 2018-05-28:
I just did a `pod install` for the LocoKit Demo App, and all the files are in the correct place and compiling correctly.

Yeah, must be a broken Pods folder or CocoaPods cache somehow.

danijelkecman, 2018-05-28:
Yep, I messed up somehow by wanting to test your updates in *develop* branch. Closing.

closed 2018-05-28 by danijelkecman

sobri909, 2018-05-28:
Heh. No worries. Glad you solved it!

The changes in the dev branch should be ready soon. The major changes are already complete. But I will hold off on sending out a stable update for quite a while, because I like to run the larger refactors through a public Arc App release as a stress test, before pushing them out as a stable pod.

danijelkecman, 2018-05-28:
Looking forward to it! :)