sobri909/LocoKit #26: Import MovesApp data

t413, 2018-07-03:
Moves just [announced](https://www.moves-app.com/) it's shutting down service all-together– you're probably about to see an influx of users. Any luck so far on perhaps importing data? I've got over 4 years worth of data I'd love to see included.

BillJJay, 2018-07-03:
I’ll second the motion. I also have over 4 years of data I want to save.

bionicl, 2018-07-04:
@sobri909 is already working on that and this feature should drop in a few days
https://twitter.com/sobri/status/1014347672621203456

sobri909, 2018-07-05:
Sorry for the slow reply. My inbox has gone insane this week! Hundreds of emails about Arc App and Moves data import 😳

I'm already working on Moves data import for Arc App right now, and hoping to have it ready by the end of the week. 

I'd also like to add some data import functionality to LocoKit itself. Ideally the standard formats, GPX, KML, etc. But step one is to quickly get it into Arc App, then look at what makes sense for open sourcing in LocoKit after that.

sobri909, 2018-07-06:
Just wanted to mention that I've made a [Patreon page](https://www.patreon.com/sobri909) for supporting Arc App and LocoKit development. 

Both Arc App and LocoKit are free to use, and the time I spend developing them is currently funded by the unrelated contract work that I periodically take on to keep the projects going. If enough people sponsor the projects, then I can instead focus 100% of my time on Arc App and LocoKit!

On a less off topic note: The Moves data importer is coming along nicely, and I'll be testing it over the weekend, and possibly submitting it to Apple over the weekend too, if all goes smoothly 😄

Thanks!

t413, 2018-07-16:
Such great news @sobri909, I can't wait! Have you given consideration to the _stationary activities_ that are a big part of moves and it's dataset? I personally have things like 'dancing' and 'ultimate frizbee' tagged on years worth of data and would like to keep going. One big (and simplifying) enhancement would be to allow those activity types to be user-defined– that way you don't have to maintain a comprehensive list of every thing from badminton to zouk.

sobri909, 2018-07-17:
> Have you given consideration to the stationary activities that are a big part of moves and it's dataset?

I'm saving them for a second later import. 

Arc App doesn't have any suitable feature to import that data into yet, so I'm just not importing it. But I'm keeping the import data ready, so that a later Arc App update can pick it up and automatically import the remaining details once there's somewhere to put it. 

> One big (and simplifying) enhancement would be to allow those activity types to be user-defined– that way you don't have to maintain a comprehensive list of every thing from badminton to zouk.

Yeah, more activity types and also custom activity types are high on my wishlist. And a bunch of people have been requesting more activity types, so it's going to happen sooner rather than later.

mikiqex, 2018-07-20:
In my case I also have a lot of non-stationary activities, like downhill skiing and golfing, plus I'm not sure if subway and tram is correctly assigned as "Train".

In imported data I keep having "Gulf of Oman" point, among correct points and routes in the middle of Europe, quite often, at least once every week, for no apparent reason. I don't have a teleport, I promise! :-D If I delete the "Gulf of Oman" point, it will probably mess up surrounding "Transport" routes.

Otherwise the import worked perfectly, you're a hero! :-)

I was thinking, would be please possible to remove the dot in the calendar for days without any timeline? Or maybe make it gray ? I wasn't using Moves for periods of time and while looking at my old data in Arc, I keep stumbling upon "empty" days. Thank you for consideration, lowest priority ofc :-)

sobri909, 2018-07-21:
> If I delete the "Gulf of Oman" point, it will probably mess up surrounding "Transport" routes.

Does the visit's location on the map match up with anywhere that you genuinely did visit? Perhaps the Moves data just has the wrong place name assigned to it. You can correct the place on it if you like, same as you would any Arc data. 

> I was thinking, would be please possible to remove the dot in the calendar for days without any timeline? Or maybe make it gray ? 

The current calendar view is a disaster. I built it in one afternoon, almost three years ago, and never meant to actually ship it in the app. What you see there is something I built for only internal testing, and never considered good enough to go in the live product 😂 I've just never had any spare time to fix it up and do it properly. 

So as soon as I get a chance, I'm going to do a *massive* overhaul of the calendar view (and search view).

mikiqex, 2018-07-21:
No, it doesn't match at all (both surrounding "Walking" activities are around "Garni Defrancesco" place):

[image]

Sorry I didn't research it deeper earlier, because the place exists in data export. Moves probably just ignored this point as incorrect, but kept it anyway, and Arc simply imported it and shows it in the timeline.

I have no idea how timeline works in Arc, but I'd say this should be easily markable as incorrect - because at 16:18 I'm in Italy, at 16:25 I'm 4800 km away in Gulf of Oman and at 16:53 I'm back in Italy.

Source data for that day (CSV+JSON) in case you're interested: [gulf-of-oman.zip](https://github.com/sobri909/LocoKit/files/2216197/gulf-of-oman.zip)