sobri909/LocoKit #27: Time at home, work and commute for the Arc App like Moves App had

adidoes, 2018-07-05:
Hi,

I know you are busy with the import feature right now but can we get this on the list of features for the Arc App? Basically have the option to set some locations as 'home' or 'work' and have the time spent there displayed somewhere in the app. Thanks!

mikiqex, 2018-07-18:
You mentioned "commute" time - I have a long tunnel on my way to/from work and therefore my trip to/from work is always broken into several timeline items. I'd love to be able to join them, at least manually (should I create a new issue for this?) Thank you.

sobri909, 2018-07-18:
@mikiqex The unmerged same-type trips bug is fixed in v2.0. Might be time to upgrade 😉

mikiqex, 2018-07-20:
Upgraded and confirmed :-)

I'd say "work" could be as a private place and on place there already are some neat statistics. I don't know if this is what @adidoes requested, but seems to me it is.

adidoes, 2018-07-20:
@mikiqex Almost, I often take lunch outside my work building and what I would like is to have all the time spent inside the building aggregated as a single number. It's the same with the commute time and time at home.

thoughtgap, 2019-03-17:
Hi,
you can get some of this info out of the data provided by my little project for analysing Arc Data: https://github.com/thoughtgap/arc-data-server

Right now you can
- mark several places as home or work
- list the commutes with `activityType` and `duration`

Right now only direct travels (e.g. home > activity > work or vice versa) can be filtered possible. If your commute is more complicated feel free to add your feedback [in this issue](https://github.com/thoughtgap/arc-data-server/issues/20)

In the future we'll plan to implement:
- a web-based analysis possibility to check durations, numbers of occurrence (e.g. heatmaps)...
- an animated map comparison of commutes

Let me know what you think!