sobri909/LocoKit #28: Make guesstimates for timeline item calorie consumption values

sobri909, 2018-07-07:
This is the second most often requested feature for Arc App, after the Moves data importer.

I was just about to add it to the tasks list in the Arc App private repo when I realised there's no reason why it can't be done in LocoKit directly.

This also relates to #24. That task is for importing this data from HealthKit. Possibly the same TimelineItem property (`item.activeEnergyBurned`)  should be used for this one. Although I suspect it might make sense to store it in a separate property and database field, so that it can be distinguished from more accurate calorie burn data sourced from wearable devices. 

## Misc Notes

- "Mifflin St Jeor Equation"
- https://en.wikipedia.org/wiki/Basal_metabolic_rate
- https://sites.google.com/site/compendiumofphysicalactivities/

bionicl, 2018-07-07:
I've done some basic digging around active burned calories values and I'll share my knowledge if that would speed up your process a bit.

Here is a MIT table I've found: https://www.topendsports.com/weight-loss/energy-met.htm
And here is the equation:

burned cal = activity multiplayer ✕ time **(minutes/60)** ✕ average speed **(km/h)** ✕ weight **(in kg)**

Activity multiplayer is based on above link, and after some math and comparing to Moves results, here are the values:
Walking: **0.79**
Cycling: **0.37**
Running: **1.03**

#### Example of 30 minute walk for 6km/h:
0.79 ✕ 0.5 ✕ 6 ✕ 80 = 189kcal

As I said, this values return almost the same results as the Moves app, but that of course doesn't mean that moves app returns correct values 🤔

Hope that'll help, good luck with development!

sobri909, 2018-07-07:
Great! Thanks 😄