sobri909/LocoKit #29: Mystery high battery usage

sobri909, 2018-07-08:
A handful of Arc App users are reporting mysteriously high battery usage. Sometimes 10 or 20 times more than expected.

My current best guess of the cause, based on user screenshots, is that for these users LocoKit is cycling between sleep and wake every few minutes, while stationary, instead of staying in sleep state until the user leaves their current location. 

My hunch is the affected devices are producing a magical rare mix of low accuracy location data and drifting location data, which could be somehow getting through LocoKitCore's filtering. Basically some poisonous combination of inaccurate location data while stationary that's slipping through the cracks and messing up sleep mode.

The problem is this never happens on any of my test devices, and is extremely difficult to properly debug remotely. So I need your help!

### Are you experiencing this bug, and are an iOS developer?

If so, I could really do with your help!

The best way to help is to install the LocoKit Demo App from this repo, and run it on the affected device in the affected location, and look for patterns.

The "LM" tab will hopefully be the most informative. Look for:

- Patterns in "Requesting accuracy" vs "Receiving accuracy"
- The number of locations in the "Latest sample"
- Incorrect changes in "Moving state"
- Drifting current location dots on the map
- A pattern of alternating short visits and short paths in the "TM" view while stationary

**Note:** You won't need an API key. I don't believe this bug is related to the ML classifiers at all.

Thanks for your help!

sobri909, 2018-07-08:
Oh, you will want to do a RELEASE build of the Demo App, not DEBUG, due to threading and inlining optimisations having a big impact on recording performance. There's a good chance the bug will express differently in DEBUG mode, and lead us astray.

nigelm, 2018-07-09:
Am seeing this.  Not currently a developer so can't unfortunately build/install the demo app.  However feel free to ping me if you need anything trying out.

Just attempted killing everything running and forcing a full restart on the device just to see if that fixes things.  The reported battery use by Arc is high, but lower than I would expect considering the effect on battery life.

Device: iPhone 6S.  Battery replaced 7 months back under warrenty by Apple.

craigeley, 2018-07-10:
Just wanted to chime in to say that I'm also seeing this on a 6S.

sobri909, 2018-07-10:
@wuhup I suspect that high mobile data usage is caused by Arc App, and will disappear once the next Arc App release goes live. (Which I'm still waiting for Apple to approve). 

My hunch is that Arc App has been inadvertently triggering a rare bug in iOS's Location Services, that in combination with some other external factors sometimes results in weirdly high mobile data usage for the affected devices. 

If my hunch about it is correct, then the next Arc App release will fix it. We'll see what happens once Apple approves it...

caramadra, 2018-07-12:
I'm also experiencing high battery & cell utilization (iPhone SE), but also not a developer.  Also willing to run tests when asked.

Thanks for the app, and for your communications when I report an issue!

nigelm, 2018-07-15:
I am still seeing this with Arc App v2.0.0 and iOS 11.4.1 on iPhone 6s over a few days.

Incidentally battery use has been high (or the battery level has gone down quickly) in the 6 hours I have been at home this afternoon.  So not completely stationary but remaining within a 15m radius - not sure if that should have triggered the App/LocoKit sleep mode.

Battery status reporting says Arc App has used 31% battery over last 24 hours.

Incidentally, please could the Arc App version number be added to the About info within the app.

sobri909, 2018-07-17:
I've just submitted an Arc App v2.1, which has further workarounds for this problem (if my hunches on its cause are correct).

I still haven't had time to investigate it directly at its source, due to rushing to get the Moves data importer finished. But now that that's out of the way, I'm going to be digging into this one properly, and getting it closed off. 

If anyone sees improved battery life in Arc App v2.1 (once it arrives tomorrow), please do let me know!

> Incidentally, please could the Arc App version number be added to the About info within the app.

Yeah, I've no idea why I haven't put that in there. Seems an odd thing to leave out. I'll add it in the next update.

sobri909, 2018-07-17:
> Incidentally battery use has been high (or the battery level has gone down quickly) in the 6 hours I have been at home this afternoon. So not completely stationary but remaining within a 15m radius - not sure if that should have triggered the App/LocoKit sleep mode.

@nigelm If your home is one of the mystery buildings that's producing the poisonous location data, then possibly it won't matter whether you're staying stationary or not, while in the building. My suspicion is that these mystery buildings are producing location data that drifts far enough outside of the phone's reported location accuracy that they're also drifting outside of LocoKit's Kalman filter limits, and the dynamic smoothing. Basically extremely nonsense data. 

I tweeted [a screenshot](https://twitter.com/sobri/status/1019147583166373889) today showing an example of what can sometimes happen, where the phone reports reasonable location accuracy (in this case, about 80 metre accuracy) but provides location data that is hundreds of metres or multiple kilometres away. 

But those events are usually rare. That particular screenshot is the first time I've seen that in this particular location in over a year. My suspicion is that the affected buildings are producing location data like that all day, without pause.

sobri909, 2018-07-17:
Omg, I just remembered the filtering feature that Arc App had for possibly this exact problem a couple of years ago. Wow, blast from the past. 

I took it out after adding the Kalman filters, because none of my test devices ever triggered the problem anymore. It seemed like the Kalmans were curing the root cause, so it wasn't needed anymore. But apparently not in all cases. 

Anyway, it's something I called (apologies for language) the "Bullshit Factor". Every time the recording engine got woken up from sleep mode by seemingly moving data, that then turned out to be nonsense shortly after, the `bullshitFactor` value was incremented by one. Basically keeping a count of the number of times the phone had made up naughty lies while stationary. 

That `bullshitFactor` value was used as a "boy who cried wolf" threshold for any future wakes from sleep. If the location data looked like moving data, but it was within the "bullshit adjusted" radius, then it would be ignored. The recording engine would assume that there was no wolf, and the phone was just making a fuss over nothing.

I wonder if it's time to put that back in.

SietseVisser, 2018-07-17:
I am having the battery drain problem on my iPhone X with the Arc app.
As suggested, I installed the demo app for LocoKit - and it works as it should, mostly sleeping when I am @home or @office, only a few % battery used. 2% for Moves, 4% for DemoApp. The 2% extra will be due to foreground usage when checking the status of the app, whereas Moves has no foreground time.

Looking forward to the Moves import, and hoping for the end of the battery drain with the next release.

sobri909, 2018-07-18:
Thanks @SietseVisser. Interesting!

> The 2% extra will be due to foreground usage when checking the status of the app, whereas Moves has no foreground time.

Yeah, foreground time in the Demo App isn't optimised at all. It does UI updates every second. Which would be an insanely bad idea for a long running app, but is useful in the Demo App, for being able to see what the data is doing in real time. 

So anyway, going by your results, it sounds like the problem isn't happening in fairly raw LocoKit on default settings. That's good! 

And also means that the problem must lie in Arc App's custom LocoKit settings. Which narrows the problem down quite a bit. 

I'm curious then to see whether Arc App v2.1 (which just went live an hour ago!) will fix the problem for you @SietseVisser. I optimised the way Arc App uses the `loco.ignoreNoLocationDataDuringWakeups` setting. That's one of the settings that Arc App changes from default, and top of the list of potential culprits. 

That setting is generally best left on its default `true` value, for battery saving reasons. But Arc App turns it off in some instances, to make it possible to detect at least some of the data of underground train trips, in cities where there's no location data at all available while on the trains. 

I guess we'll know in a day or two, once v2.1 has had at least 24 hours in the wild!

caramadra, 2018-07-18:
Not seeing a version # under "About..."  on my iPhone.  Where do I find it?  Thanks.

sobri909, 2018-07-19:
> It is at 7% while moves was at 6%. I used Arc 15 minutes in foreground, and Moves 2 minutes in foreground. So I am happy.

Yay! 😄🎉

That's basically what I see on all my test devices. The battery consumption is either the same as or less than Moves, when given equal screen time. So it sounds like the problem is solved for you 😄

I think it'll take another day or two to know for sure if it's solved for everyone. The Moves importer is not well optimised, because I was rushing to get it shipped, so it uses a lot of CPU, which will make battery percentage comparisons unreliable for another day or two.

> Question about Moves import, what happens if I import all data, thereby overlapping with the days I used Arc also? What happens to the overlap in data?

The importer will skip over any overlapping data and not import it. 

The original plan was to import it all, then let LocoKit's processing engine merge it all together automatically. But that was awful. Unusable mess. Because Moves data is all rounded to the nearest second, while LocoKit is storing millisecond precision. So the merged paths were unusable messes. 

So yeah, I didn't do the merging! Instead it'll just import Moves data from before your Arc data.

sobri909, 2018-07-19:
@caramadra Yeah, I need to add the version number to the About view. That's on my todo list. 

In the meantime you can find it in the App Store, if you search for "arc app" then see whether it shows an update button or not.

caramadra, 2018-07-20:
I updated yesterday.  Still seems like really high use (fully charged phone down to 25% battery by 1pm).  Any adjustments I should be or can be making?

craigeley, 2018-07-20:
I am also still seeing this. From the battery screen, Arc is showing very high background usage—like 12 or 13 hours in the past 24. I imagine that some of this is from me having deleted the app and then reinstalling it, which triggers a backup from iCloud that can't seem to be adjusted or stopped. When I woke up this morning, that backup was complete, but my battery was still getting crushed. 

Also, if the app switches to "sleep" or "deep sleep" mode when I am home, should I be seeing so many data points in and around my house?

[image]

Finally, as you can see in the screen shot, I have bad cell service at home, so I have wi-fi calling turned on. Not sure if that might be a part of this narrative, but just trying to run through possible oddities in my device / use case...

caramadra, 2018-07-24:
Still experiencing the high battery usage.  Let me know if there's anything I (a non-developer) can provide to help you debug.  Also have a question and comment:  What's the difference between a red ellipsis (...) and a gray/black one?  If it's correct, do I still need to tap / edit it to improve naming?  (Okay, _two_ questions!)  And there seems to be a lot more "thinking" going on, since the last update, so don't know if that's why my battery use is high.

Thank you!  Six more days until Arc becomes my full-time, go-to diary/tracker!

mikiqex, 2018-07-25:
Similar here (iPhone SE), 4 min on screen and 4,6 hrs background activity (and latest Arc) = 21 %.

I can't tell how much of that percentage is due to the on-screen time, but overall it's OK by me if on-screen battery usage is high, as I can control it. But I was at home all day and Arc took 6 % of battery (doing bascially _nothing_, I'd say :-). Sorry, I forgot background activity time, but on-screen was zero.

I'm wondering how would affect battery life if CLLocationAccuracy is set to 100, as I value battery over precision. Also, Is Arc doing anything else beyond recording in the background?

Thank you.

sobri909, 2018-07-25:
@mikiqex Do you have the v2.1.3 update yet? That's the one that fixed the iPhone SE bug. 

The iPhone SE bug was the one causing buttons to be untappable, and was also causing high battery drain while the app was in the foreground. So the fix should make foreground time much less battery expensive for you now!

> But I was at home all day and Arc took 6 % of battery (doing bascially nothing, I'd say :-)

If the phone was plugged into power and wifi, then it was probably doing its daily iCloud backup, and possibly some ML model updates (local / transport knowledge). Those things are only done while plugged into power, to avoid battery drain, but the Settings -> Battery view doesn't distinguish between energy use while on power or battery, so it can be a bit misleading in those cases. 

> I'm wondering how would affect battery life if CLLocationAccuracy is set to 100, as I value battery over precision. Also, Is Arc doing anything else beyond recording in the background?

LocoKit already does this 😉

LocoKit dynamically adjusts the `desiredAccuracy` depending on whether you're stationary or moving, indoors or outdoors, and depending on the peak achievable accuracy in the area (so that it doesn't waste energy requesting an impossible level of accuracy). It's basically adjusting the requested accuracy every few minutes to match the conditions, to get the highest accuracy with the least battery drain. 

The location recording is actually the least battery expensive part of Arc App! (When everything is working correctly - this Github task is for finding and fixing the edge cases where it's not working properly).

When LocoKit's location recording is working correctly, an iPhone SE should be able to record for 24+ hours on a single charge, with Arc App often showing as using about the same amount of battery as the "Home & Lock Screen", if there hasn't been any screen time.

My iPhone 7 test device (with a nearly dead "needs service" battery) gets 30+ hours recording on a single charge. And the Plus model phones should be able to get 2-3 days on a single charge. 

The biggest unfixable battery drain, when all else is working correctly, is the map view. The screen time, CPU time, and GPU time for the map view is unavoidably expensive, unfortunately.

sobri909, 2018-07-25:
@caramadra You're on an iPhone SE too yeah? In that case hopefully yesterday's v2.1.3 update will fix the problem for you too! Let me know how you get on.

mikiqex, 2018-07-25:
@sobri909 Ok, I have v2.1.3 since today, so my battery report may be obsolete... thanks **so much** for fixing untappable buttons!

Also thank you for your patient explanation how (cleverly) things work in Arc. I really appreciate it.

If iPhone calculates into battery time even operations made while charging, the true impact on battery life is hard to tell. And about the map view - this may explain why there is no map in timeline view in Moves :smile:

sobri909, 2018-07-25:
No worries. Thanks for hanging in there 😄

> If iPhone calculates into battery time even operations made while charging, the true impact on battery life is hard to tell.

Yeah, it's frustrating, because the Settings > Battery view can be very misleading, and give the wrong impression (both good and bad) a lot of the time.

I used to have Arc App updating its iCloud backups as often as twice a day, if the phone was plugged into power and wifi often enough (eg both at home and at work). But even though that had zero impact on battery life, it did have a big impact on the battery percentages in Settings > Battery. So had to change it to only once a day, to avoid having to explain the numbers all the time 😞

> And about the map view - this may explain why there is no map in timeline view in Moves 😄

Yeah, Moves made a smart decision by not putting a map in the main UI 😂 It gives Moves a perceptual advantage in Settings > Battery a lot of the time. Especially so if the user has spent a lot of time browsing around in Arc App. All that map view zooming and updating can get expensive, and Moves doesn't show or update maps anywhere near as often.

caramadra, 2018-07-25:
@sobri909 , Yes, I'm on an SE.  I don't think that the update helped me, which makes me wonder if something else is going on.  I get, from your post, that the battery stats are not a great indicator, but they're all I have to go on and, if I'm having to charge my battery twice a day, I've got a problem.  I'm having Arc in the foreground maybe two or three times a day, for a few minutes, to update the locations in my timeline.  And the (beta) Battery Health indicator has dropped from 99% to 97% max capacity on my 2+ year old SE just since I've been using Arc.

Are others with an SE and ios 11.2.something seeing improvement?

sobri909, 2018-07-26:
@caramadra Yeah, you shouldn't have to charge more than once a day. Something ain't right.

My iPhone SE test device ends each day with more than 50% battery remaining, on the days I do battery testing on it.

What are the numbers in the Battery > Settings view showing now? As you say, they're all we've got to go on. 

> Are others with an SE and ios 11.2.something seeing improvement?

Oh you're still on 11.2? I recommend updating to the latest 11.4. It possibly won't make a difference to the battery performance, but it's worth eliminating it as a possibility.

nigelm, 2018-07-26:
General status update - basically no change...

iPhone 6S, iOS 11.4.1, Arc: 2.1.3

Still seeing battery life much lower than before Arc (ie not making it to end of day where I used to have pushing 50% left).  Occaisional instances where phone is unaccountably warm.  Battery left percentage is erratic, sometimes dropping multiple percent per minute, recovering, diving...

Arc sometimes dies overnight (while on charge).  Had wondered if this was due to it fighting with Pillow (sleep monitoring app) - so have not had that running recently, and Arc still went dead last night (this was on 2.1.3 as well, so bug still there).

Battery usage (from Prefs), typically around 35% during the day, Currently (first thing, been on charge overnight) showing 15%

sobri909, 2018-07-28:
Hi @nigelm. Sorry for slow reply. I've been absolutely swamped lately. Still getting hundreds of emails since the Facebook announcement, and I can't keep up. 

I've just had another report of an iPhone 6s user getting high battery drain. I wonder if there's something specific about that model. It also happens to be the one iPhone generation that I don't have a test device for. (I have iPhone 5s, SE, 6, 7, X test devices, so I guess you could also say I don't have an 8).

If it does turn out to be something 6s specific, I'll go out and get one, to add to my test devices. 

Anyway, are you able to run the LocoKit Demo App on your phone? If so, it would help greatly if you could try that! That will help to determine whether the problem is in LocoKit, or is Arc App specific. 

Make sure to build and run the Demo App under the RELEASE scheme, but DEBUG, as the compiler optimisations have a massive impact on energy consumption.

Thanks 😄

caramadra, 2018-08-05:
I upgraded my iOS, then started my moves import - then removed my phone from its case, as it was getting very warm!   Any way to pause the import, if I need to use my phone on battery tonight?   (The import dropped my battery 20% in about 5 minutes!)

You get the battery drain resolved, and I’ll be a backer!

Thanks

sobri909, 2018-08-06:
Hi @caramadra! The Moves importer tells you to plug into power, for that exact reason 😉 

The Moves importer is not optimised for battery use, it is attempting to get the job done as quickly as possible. 

Once the import has completed, Arc will also need to do a full backup of the imported data (unless you have turned off iCloud backups). So you should expect the Settings -> Battery view to show high battery use for Arc for the next 1-3 days. 

However, this is misleading. The Settings -> Battery view does not distinguish between energy use while on battery and on power. Arc App only does its backups while on power, so they can never drain the battery. 

This task if for dealing with mystery battery drain caused by irregular location data inside some buildings. So it doesn't deal with issues from the Moves import, which is a one time event, and is meant to be done while plugged into power 😉

@caramadra, once the Moves import has completed, and you have given the backup a day or two of grace to complete, please check the numbers in the Battery > Settings view, taking into account screen time relative to other apps. A screenshot would be very useful!

As far as I can tell, the battery drain issue has been resolved for effectively everyone, with only a few rare cases that might be showing a possibly related issue. The emails I'm receiving that show excessive battery drain now are either new users in low "local knowledge" areas, users with acceptably high battery use due to high screen time, or users who have been swiping the app closed (which will result in increased battery use). I'm receiving very few, if any, emails suggesting that the issue inside some buildings might still be present.

So to help you debug the problem you're having, I will need more information! 

> You get the battery drain resolved, and I’ll be a backer!

Thanks for the thought 😊 But what Arc App really needs is backers **now**. If you want the app to continue, it is important to support the project now. If it does not meet its funding target, it will disappear.

craigeley, 2018-08-13:
@sobri909 — did you see my [old comment](https://github.com/sobri909/LocoKit/issues/29#issuecomment-406701729) about the data points being generated around my house?

Also, like @caramadra, I'm using a 6S and you have stated that you do not have a 6S testing device, correct? So it seems like the battery drain issues have possibly not been resolved on that device, or else @caramadra and I have the same rare case.

craigeley, 2018-08-14:
@sobri909 Cool—latest app store build seems to solve everything for me!

sobri909, 2018-08-15:
@craigeley Oh wow, cool! I didn't expect that. 

Arc App v2.1.9 did include a fix for a case where Arc might try to make excessive, unnecessary "wake up call" requests to the server. A bug that was introduced I think around v2.1.0.

I haven't seen any energy consumption differences either before, during, or after that bug. So I'd assumed that it wasn't doing any harm. But I wonder if under some conditions it was stacking up requests, and chewing through energy.

Aside: The wakeup call requests were added as another way to make sure Arc stays alive (or rather, gets restarted) in the background, if iOS ever decides to kill it. They're requests for a "silent notification" from the LocoKit server, timed to fire at either the predicted leaving time for the current visit, or in one hour, if no predicted leaving time is available.

I suspect the original problem for this task still exists though. We've just narrowed down the cases of excessive battery consumption a bit more! Narrowing in on the target, on the most difficult to solve problem, by chipping away at the easier edge cases.

craigeley, 2018-08-15:
To be honest, there are a LOT of factors at play here—I probably missed a few updates because I uninstalled the app; I’m running the iOS 12 beta, etc. So not sure it was exactly 2.1.9 that fixed things for me. But hey! I’ll take it!

fishtopher, 2018-08-16:
Hello,
I still get pretty high usage, enough that if I know I won’t go near a charger I turn arc off for the day :(

Right now it’s at about 20% on the battery screen (1m on screen, 21h bg).

I’ll try install the LKdemo at the weekend if I can, but wanted to add and then data point for you:)

(Im also on an iPhone 6S..)

nigelm, 2018-08-16:
Not really had anything new to add (and I have been away for a while) - still seeing high usage and some slight weird effects (like the attached one from last night).

Before I went away I attempted to build the Demo App but failed at that time as I don't normally work with the Apple tools (or this language or...).  I think I was missing a dependancy.  Will likely try again at the weekend.

[image]

caramadra, 2018-08-28:
> Oh you're still on 11.2? I recommend updating to the latest 11.4. It possibly won't make a difference to the battery performance, but it's worth eliminating it as a possibility.

11.4 update didn't help.  Arc is still my biggest battery user (according to the usage stats, anyway), and I still have to charge at least once during the day to ensure that I don't run out of battery while I'm away from charging sources.

Any other ideas?  I'd like to help you as a backer, but I need to see that I'm going to be able to use the app.  

For today, I've shut off tracking.  Battery's still dropping, so I'm considering deleting some other apps that are high in the list, to see if there's some kind of cross-talk making it look like you're using more than you are.

sobri909, 2018-08-30:
> Any other ideas? I'd like to help you as a backer, but I need to see that I'm going to be able to use the app.

@caramadra I think we need to keep discussion of money out of this.

A single backer cannot fund the development of bug fixes, maintenance, or new features. These sorts of tasks cost thousands or tens of thousands of dollars of developer time.

So please be aware that when you make statements like that you are waving $5 in front of my face and saying "maybe you can have this!", which is the equivalent to waving small treats in front of a performing animal. It is insulting. 

If you want to support the project, that's great! If you don't want to support it, that's fine! But please don't wave lunch money in front of me as though it's something I need to bark and beg for. That will not be well received.

So yeah, let's please leave money out of this discussion. This Github repo is for discussion of technical issues.

---

Back to the technical issue at hand: I haven't received much in the way of identifiable debug information on this issue in quite a while. A small number of people are still experiencing higher than expected battery consumption, but the obvious patterns are either no longer present, or I'm not receiving much evidence of them. To be able to move forward for the remaining cases, there needs to be something to work with, something to debug.

@nigelm's screenshot is a good example. That shows a case where it appears as though the location data is drifting, producing human-like movements, even though the phone is stationary. If that kind of pattern can be identified in line with increased battery consumption during that period, then that's something to go on! That would suggest that the original posited problem is still present in some cases.

caramadra, 2018-08-30:
I’m not sure how to respond to this.  You are the one asking me to become a backer, thru the pop ups in the app.  I certainly didn’t imagine that addressing that request would be insulting.  I’m simply trying to get across to you that I can’t justify paying for an app if I will be unable to use it, though I want to use it, and back it.  (And please know that $5/month [not just $5] is not insignificant to my family . )

That said, the point of my “Any other ideas?” Was to try to find out what troubleshooting I could do or what information I could provide that might help you pin down why the use of Arc seems to be sucking the charge from my battery.   

I had it off today, started at 7am with 100 % and made it to 5pm with 30% left.   That’s still not great but troubleshooting with Apple yesterday indicates that my battery is “healthy.”   I’m removing apps from my phone in order of their power usage in an effort to see what it takes to get through a full day.  If you want screen shots, tell me what to look for.  

I move a lot in a small area for 8 hours.  I thought those “drifting” entries were accurately reflecting that.  I am not usually able to watch for a relationship between those and drops in available power, because I’m usually intent on another task, during the day.  I will continue to eliminate apps to see if I can get my phone thru a day without needing to charge, then maybe I can see more of a direct relationship between Arc and the usage.

sobri909, 2018-08-30:
> I’m not sure how to respond to this.  You are the one asking me to become a backer, thru the pop ups in the app.  I certainly didn’t imagine that addressing that request would be insulting.

When it is implied that money is being withheld until some task is performed, and when the money being discussed is a tiny fraction of the cost of the task in question, that is when it becomes insulting. In those cases it is better to not mention it at all. 

For technical discussions, bug hunting, feature requests, etc, it is not important whether someone is supporting the project or not. Everyone's requests are treated equally, regardless of whether or not they supporting the project. (And I would prefer to *not* know who is and is not supporting the project, so that that fairness can remain unbiased). 

So yes, it would be greatly preferable to keep money out of this discussion. If you wish to discuss that subject further, then I'm always available via email at matt@bigpaua.com. 

--- 

> If you want screen shots, tell me what to look for.

The first step is to identify the times when battery drain is excessive. It will be unlikely (or extremely unlucky) for the battery drain to be consistently high across the entire day. There will be specific times during the day when energy consumption is higher, and possibly unreasonably high. 

Once those times can be identified, then we can look for patterns in the data, patterns in app usage, etc, to give us clues what's causing the excessive drain. 

> I’m removing apps from my phone in order of their power usage in an effort to see what it takes to get through a full day.

The Settings -> Battery view won't tell you about which times of the day are problem times, but it will give you comparative values between apps, that can be useful in determining which apps are using reasonable or excessive amounts in comparison.

In the Settings -> Battery view, the most important number to look for is the "screen time", and then to compare between apps with similar amounts of screen time, to see whether one is using a significantly higher percentage of battery than the other. If two apps have the same screen time and same percentage of battery drain, it's likely that the amount of drain is normal and not excessive. But if one is significantly higher percentage of battery for the same amount of screen time, then that can reasonably clearly point the finger at that app as using too much.

> I had it off today, started at 7am with 100 % and made it to 5pm with 30% left.

Draining 70% of the battery in 10 hours may be normal, if there is a lot of screen time. iPhone batteries are only capable of a limited number of hours of screen time. If the combined screen time of the apps used during that time add up to many hours, then that would look like a normal amount of battery drain. But if the screen time is less than an hour or two during those 10 hours, then yeah, something isn't right.

fishtopher, 2018-09-01:
Took my time getting here, but just trying to build & install the app now.  I'm having some problems though - I've not actually done any ios dev for years, and never used swift/cocoapods so not sure what I need to do vs what should be happening automagically...

So far I've managed to clone the project, gem install cocoapods, pod install - however now I get these errors:

```
diff: /Podfile.lock: No such file or directory
diff: /Manifest.lock: No such file or directory
error: The sandbox is not in sync with the Podfile.lock. Run 'pod install' or update your CocoaPods installation.
```

stack overflow suggests there might be something missing in a pods directory?  However I can't find a pods directory...

```
Pod installation complete! There are 6 dependencies from the Podfile and 8 total pods installed.

$ ls

ActivityTypeClassifierExamples.md	LocationFilteringExamples.md		LocoKitCore.framework			Screenshots
ArcKit.podspec				LocoKit					LocoKitCore.podspec			TimelineItemDescription.md
ArcKitCore.podspec			LocoKit Demo App			Podfile					docs
BackgroundLocationMonitoring.md		LocoKit Demo App.xcodeproj		Podfile.lock
CHANGELOG.md				LocoKit Demo App.xcworkspace		Pods
LICENSE					LocoKit.podspec				README.md```

Any ideas / suggestions on what to do?

sobri909, 2018-09-03:
@fishtopher Hmm. I can see the Pods dir in your dirs list there... 

You could try deleting the Pods dir then doing a fresh pod install. If anything ever goes weird with pod install, that's an easy (if brutal) way of fixing it up. 

Which branch are you using? I'm working purely in the develop branch at the moment, so I haven't touched master in quite a while. Although master was all installing and building fine at release time.

nigelm, 2018-09-20:
Early days... but it looks like overall battery life on my iPhone 6s has got significantly better following update to iOS 12.

*Disclaimer: potentially spurious as its less than two full days without a ton of upgrade etc messing. 
 Also not tried witout Arc running...*

sobri909, 2018-09-20:
@nigelm That would be some welcome news for iOS 12! So far I've found iOS 12 to be an unpleasant step backwards for Arc / LocoKit. 

I'm seeing much higher rates of background fails in iOS 12 (in Arc's analytics, and on my test devices). And much higher rates of nonsense location data, with some devices "sleepwalking" on almost a daily basis, several kilometres away from where they really are.

On battery life, I haven't been able to do useful comparisons with iOS 12 so far, because none of my test devices experience high battery consumption. But they certainly are all suffering from lowered reliability and accuracy on iOS 12. Hopefully Apple will be cleaning up those rough edges in the first patch release.

EugeneGoloboyar, 2018-09-25:
I've played a little with SDK (latest version, master branch).
I'm testing it on my project which has NetworkLogger. (I use it to inspect network request, log specific info)  
What I've found strange that my application using SDK executes tones of API requests for the same coordinates. I tried to change my location just by walking at the office. 
So I don't have significant location changes, just stationary.
Also I use default accuracy options (30 meters).
Suppose it would be helpful for this issue debugging.

[image]

iOS version: 11.4.1
Device: iPhone 6+ 
Xcode: 9.4.1
GPS/Wifi/CoreMotion is Enabled.

sobri909, 2018-09-25:
Hi @EugeneGoloboyar! It would be better to file a separate issue for that, as it isn't related to the problem of high battery consumption while stationary. Thanks 😄

EugeneGoloboyar, 2018-09-25:
Ok, I've created it as a separate issue. Hope you have ideas how to enhance this)
https://github.com/sobri909/LocoKit/issues/45

closed 2018-11-04 by sobri909