sobri909/LocoKit #32: Gpx month export sometimes doesn't include movement type (Arc app)

bionicl, 2018-07-22:
I have one day fully corrected without any 'low confidence' errors and it features many car trips.
When I exported it as only one-day GPX everything was fine, but full-month GPX file lacks many of type attributes (`<type>car</type>`).

Checked that with a few months and every time it looks the same.

sobri909, 2018-07-23:
Hm. Interesting! I'll have a poke around in the GPX code soon.

I also want to get it out of Arc App and open sourced into LocoKit soon (#18). Which will make it easier for other people to spot my mistakes like this.

sobri909, 2018-08-19:
I finally found the cause of this!

1. https://github.com/sobri909/LocoKit/blob/develop/LocoKit/Base/Timelines/Items/ItemSegment.swift#L130-L132

2. https://github.com/sobri909/LocoKit/blob/develop/LocoKit/Base/LocomotionSample.swift#L157-L160

If the first sample in the segment doesn’t have a confirmed type, and the classifier hasn’t been re-run on it since the last time the sample was loaded from the db, it’ll have a nil activityType. 

The fix will be to make sure the classifier has been run over all unconfirmed samples before building the GPX (and JSON).