sobri909/LocoKit #39: Defer locations to max of one per second

sobri909, 2018-08-30:
I've had a report of Arc App using excessive battery when running alongside Google Maps when Google Maps is providing directions. 

My suspicion is that Google Maps is requesting the highest level of location accuracy, which will result in Arc / LocoKit also receiving that level of accuracy, and *also* that frequency of location updates.

Which means instead of location update frequency maxing out at about 1 Hz, it might be going as high as 2 Hz. Which is completely unnecessary for LocoKit, and will be doubling the energy cost of the filtering layers.

So yeah, idea is to either use deferred location updates, to set a maximum sampling frequency, or to simply discard locations that arrive too soon after the previous.

thoughtgap, 2018-08-30:
My thoughts on this:
Back in the day with Moves, I used this trick to make the Moves track more accurate. On long bus rides through the countryside for example, I'd have a long line without any detail with just Moves. If I was running a GPX-Recorder or Navigation app at the same time, the Moves track would be very detailed.
This came (and I'm assuming this) at now extra battery cost for Moves App.

I would appreciate if Arc wouldn't ditch any information that'd make the tracking more accurate.

sobri909, 2018-09-10:
So the CLLocationManager "deferred locations" feature can't be used. It requires the desiredAccuracy setting to be set unnecessarily high (ie requesting `kCLLocationAccuracyBest` or `kCLLocationAccuracyBestForNavigation`). 

Instead I've just made the LocomotionManager ignore incoming locations if they arrive too soon and aren't of higher accuracy than the previous.

closed 2018-09-10 by sobri909