sobri909/LocoKit #40: ArcApp merges few transport types into one

bionicl, 2018-09-01:
So basically Arc app (newest version) started merging some of different transport types into one. For example:
> 1km walk
> 5km train
> 2km walk
Merges into "8km walk". When I go to 'Recorded activities' tab and I try to split activities, it detaches for a few seconds and then merges again.

I only had such problems with data from Moves import, demonstrated on the video: (might be a bit confusing without finger circles, but I hope it'll help!)

[ScreenRecording_08-31-2018 22-18-53.mp4.zip](https://github.com/sobri909/LocoKit/files/2342342/ScreenRecording_08-31-2018.22-18-53.mp4.zip)

bionicl, 2018-09-01:
Trying to split by clicking 'walking' (in the case of this video) doesn't do anything different.

One experiment I tried, was changing 'train' to different type I haven't used before - for example 'boat' - it was successful! But changing back to train/car caused the same as in the video. I had a few same examples, each one behaved the same.

I'll check in the free time if there is something unusual in moves json day files, but I don't think that's the case

sobri909, 2018-09-01:
Wow. If it acted differently when assigned to boat, then ... I'm very confused. The assigned type shouldn't make any difference, as long as it's not the same as the type as the item it's being split out from.

The processing engine always reprocesses after any manual change, thus reapplying it's own rules. It's not allowed to override our manual ActivityType and Place assignments, but it will reassert control over TimelineItem boundaries and sample/segment ownership. So when an ItemSegment is split out, the processing engine might decide that the new arrangement isn't workable, and either partially of fully merge things back together again. 

But in the case in your video... I just can't see how it would believe that's the right thing to do. It doesn't fit with any of the rules. Either there's something unacceptably wrong with the data that we're not seeing, or there's a weird bug / logic fail in the processing engine rules. The video really doesn't show any hints of problems with the data, so it must be a processing engine rule failure somehow. But... how? Heh. Perplexing.

sobri909, 2018-09-01:
For a path-path merge, I'd normally expect the answer to be in here: 

https://github.com/sobri909/LocoKit/blob/develop/LocoKit/Base/Timelines/MergeScores.swift#L114-L150

But in this case none of those rules look like they could possibly have been triggered. And the fact that the merge happens in stages, rather than a single act, means that it must surely be a visit edge cleanse. So maybe the answer lies in a visit item on one of the ends of the resulting mega path. Something fishy going on in the visit's data, perhaps.

closed 2018-11-04 by sobri909