sobri909/LocoKit #47: I think  LGPL is conflicting with AppStore rules

sprhawk, 2018-10-08:
https://forums.developer.apple.com/thread/27709

So an LGPL library should not be used (legally) on AppStore

And because app in AppStore is not allowed to be used dynamic link,  the app with an LGPL lib should be considered as GPL. 

What do you think?

sobri909, 2018-10-08:
LGPL is designed specifically for linking libraries to products with different licenses. Dynamic or static is irrelevant. It is specifically the purpose of the LGPL (as compared to the GPL).

Of course you could argue the finer details of the legal technicalities endlessly if you like. People literally have, for far too many years. But it is purely wasted time to do so. Picking over such details will result in nothing in terms of legally actionable outcomes, and anyone would be mad to try to take any of them to court. 

If you want to read up on the decades of bickering about it, google searches will guide you through 😉 But I'm not interested in hashing out those debates for the millionth time here.

Good luck in your searching!

closed 2018-10-08 by sobri909

sobri909, 2018-10-08:
https://en.wikipedia.org/wiki/GNU_Lesser_General_Public_License#Differences_from_the_GPL

That summarises the linking specifics of the LGPL reasonably well. 

Please be careful about what you read and believe on the internet on this topic. There is a large amount of misinformation, and much of it intentionally misleading, for political purposes.

Read the license directly yourself, if you are concerned. And read authoritative websites on open source licenses. The license is easy to read, and authoritative open source license websites will explain its meaning clearly.

sprhawk, 2018-10-08:
Sorry for missing statement for statically linked program: 

> Alternatively, a statically linked library is allowed if either source code or linkable object files are provided

But I still question LGPL whether is not allowed on AppStore due to above restriction

sobri909, 2018-10-08:
That's the "finer details of the legal technicalities" that I mentioned above. It's a situation that literally no one cares to explore. No one wants to do that. It would never be applied. It was advice that was written decades ago, for a situation that doesn't make any sense for iOS apps. 

Digging down into that stuff is how you waste time and achieve nothing. It won't ever legally affect you or your project or my project or anyone else's project.

sobri909, 2018-10-08:
The point is, if someone legit came to you and asked you to fulfil that part of the license, you could provide them with binary objects for them to relink against, and build their own app with their own linking against a replacement LocoKit, painfully, by hand, and everyone's time would be wasted and no one would have achieved anything. 

It's just not going to happen. The letter of the license can be adhered to for App Store apps. That's no problem. But no one is ever going to test that aspect of the license, because it would be a humungous waste of time for absolutely everyone involved, including the person requesting it.

Like I said, this is all a dead end that it's best not to waste time or thought on.