sobri909/LocoKit #49: Driving mode can't be detected on iPhone 6 (iOS 11)

mustafiz012, 2018-10-30:
**Tried on motor-bike**

**Log** says, **.movingStateChanged** (**moving** or **stationary** or **visit**)
--> when moving, it's always on **'walking'** mode

If you please make this issue clear to me. Is this a **Device issue** or **iOS** version issue something related to your side? 

**NB:** I've used this commit, [https://github.com/sobri909/LocoKit/commit/f65b00eeb2d17dfb5d52a636ad732a2781d1d2a0](url)

sobri909, 2018-10-31:
Hi @mustafiz012!

The first thing to check is to make sure you're actually using LocoKit's activity type classifiers. The default setup in the LocoKit Demo App doesn't use the classifiers, because they require an API key (in the init method of the main ViewController). 

Once you've ensured that you're definitely using the classifiers, have a look in the "AC" tab. You should see a "Region coverageScore", and then a list of the activity types that the classifier currently has available, sorted by their probability score for the most recent sample. 

If "Car" is not shown in the list of types in those results, then it means there is not enough coverage data in that area, so it will not be possible for the classifier to identify that transport type specifically. 

In cases where there isn't enough coverage data, the best option is to look at the score for the top result, then fall back to showing the user a generic result of "transport" when the top scoring type has a probability score that is below a hardcoded threshold (eg below 0.001, or some such). You will want to tweak that threshold until you get results that feel appropriate for your own app, as different apps will create different user expectations.

Not that the "Core Motion activity" shown on the "LM" tab is not from Arc, and is instead the phone's built in Core Motion types. Core Motion types are: stationary, walking, running, cycling, transport, and are detected with lower accuracy than LocoKit's own types (thus the need for LocoKit to provide classifiers, given that iOS's built in classifier is not capable of high accuracy detection).

mustafiz012, 2018-10-31:
So if I create an API Key and assign like this, **LocoKitService.apiKey = "API_KEY"** in LocoKit Demo app's ViewController, it should work?? The docs described as below,

**Note that if you are using a TimelineManager, activity type classifying is already handled for you by the manager, on both the sample and timeline item levels.**

sobri909, 2018-10-31:
Yep! All you need to do is provide an API key.

You do also need to provide the recorder with an activity types data source, but that's already done for you in the Demo App code, on the next lines after the setting of the API key. So as long as you set a valid API key, the rest will be taken care of for you.

mustafiz012, 2018-10-31:
That's great!! It shows 'Motorcycle' when I ride a bike after integrating the API_KEY.
One thing I need to be clear, shouldn't I get at least 'automotive' or 'transport' mode without API_KEY?

sobri909, 2018-10-31:
> That's great!! It shows 'Motorcycle' when I ride a bike after integrating the API_KEY.

So you were on a bicycle? That sounds like the wrong result then 😉

But as long as that's only on a per sample basis, and not for the final TimelineItem, that's nothing to worry about. LocoKit's TimelineItems (ie Visit and Path objects) provide a `modeActivityType` property, so that you can know what was the most common type of item's samples. So if 51% of the samples classified as "cycling" and 49% classified as "motorcycle" then the Path's `modeActivityType` will be "cycling".

There's no chance of ever achieving 100% accuracy in type classification, so some misclassified samples is expected and normal. The only time that single sample classification mistakes are a worry is when they are more frequent than the correct type, which should in most cases be rare. 

Oh, the TimelineItem objects actually provide two mode type properties. There's `modeActivityType` and `modeMovingActivityType`. The latter simply excludes the possibility of "stationary", thus selecting the next highest scoring type. So that one is useful for cases where you're confident that the activity must have been a moving activity of some sort, and can thus safely exclude the possibility of "stationary" being the correct answer. 

> One thing I need to be clear, shouldn't I get at least 'automotive' or 'transport' mode without API_KEY?

Without an API key, the only type you will see will be iOS's built in [Core Motion activity type](https://developer.apple.com/documentation/coremotion/cmmotionactivity), which is provided by the phone's M co-processor.

The accuracy of those is fairly low, so they're not useful for anything but the most superficial tasks. But LocoKit still collates and provides them on each LocomotionSample, for convenience sake. Personally I wouldn't bother using those for anything user facing, as they're more likely to be wrong than right. But they do offer an interesting data point while doing debugging, sometimes. And LocoKit's collection of them has near to zero energy cost, so recording them and adding them to LocomotionSamples is harmless enough.

closed 2018-11-01 by sobri909