sobri909/LocoKit #51: Delete/manage private places.

jandamm, 2018-11-10:
I accidentally created a private place and noticed that there seems to be no way to delete a private place.
Would be nice to remove the wrong private place. Now I just had to rename it to "wrong entry".

sobri909, 2018-11-11:
Hi @jandamm! Arc will delete private places automatically once they are no longer assigned to any visits. 

Search for the place name, to find all remaining visits to it, then reassign those to their correct places. Shortly after the private place has no assigned visits remaining, Arc will delete it. 

Hope that helps!

Thanks 😄

jandamm, 2018-11-11:
Thanks for the answer. It hasn't deleted itself by now. I'll just wait a bit longer.
Cheers

sobri909, 2018-11-11:
If it hasn't been deleted, then either a visit is still assigned to it, or somehow the Place model update failed to happen. 

If the former case, then best to check with the search view again, to make sure there's definitely no visits still assigned to it. 

If the latter case, you can trigger another Place model update by assigning a visit to that place, then assigning it back to its correct place a few seconds afterwards. (Each time a visit's place is changed, it queues up a model update for both the previously assigned place and the newly assigned place).

jandamm, 2018-11-12:
Figured it out now.
It was still assigned. 

Apparently deleting a visit does not remove the assignment to a private place.

sobri909, 2018-11-12:
Oh, good catch! Yeah that sounds like a bug.

Deleting a visit will cause the visit to be merged into an adjacent item. And will probably not trigger a place ML model update. So that leaves a loophole where an unused place can continue to exist and not get deleted automatically. 

I'll file a task for it in Arc's private repo, to make sure I get it fixed over in that repo. (Arc places ML engine isn't open sourced in LocoKit, due to not wanting to step on Foursquare's toes).

Thanks 😄

closed 2018-11-12 by sobri909