sobri909/LocoKit #52: XCode 10.1 & Swift 4.2 can't Build

Faydee1220, 2018-12-06:
I already try 
pod 'LocoKit'
which is build with Swift 4.1 and version is 5.2.0, this one can't build
Then I try
pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
which is Swift 4.2 and version is 7.0.0, but still got 3 errors:
Pods/LocoKit/LocoKit/Base/LocomotionSample.swift:27:30: Use of undeclared type 'ActivityTypeTrainable'
Pods/LocoKit/LocoKit/Base/LocomotionSample.swift:113:35: Use of undeclared type 'ClassifierResults'
Pods/LocoKit/LocoKit/Base/LocomotionSample.swift:145:9: Use of unresolved identifier 'CLPlacemarkCache'

Any solution? I find related issue like https://github.com/sobri909/LocoKit/issues/20 but still no help.

Faydee1220, 2018-12-06:
OK, I solved, because I miss pod 'LocoKit/Timelines'. The full pod is:
pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKit/Timelines', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'

closed 2018-12-06 by Faydee1220

sobri909, 2018-12-06:
Glad to see you figured it out 😄🎉

I'll be merging develop back up to master soon (hopefully). Then this bit of awkwardness won't be necessary, and life will go back to normally versioned releases. 

It's just a matter of missing documentation that's holding back that release process at the moment. Sigh. Always more work to do!

sobri909, 2018-12-06:
Actually, just looking at this again, it looks like you've exposed a bug there. It looks like the base LocoKit spec has dependencies on the `Timelines` spec, which is not supposed to be the case. I'll get that fixed today.

poulpix, 2019-02-10:
> Actually, just looking at this again, it looks like you've exposed a bug there. It looks like the base LocoKit spec has dependencies on the `Timelines` spec, which is not supposed to be the case. I'll get that fixed today.

Did you manage to fix it yet? It looks like it is still the case, as when doing a `pod update` today with only `LocoKit` and `LocoKitCore` referenced in my `Podfile`, I got the same missing types, and also had to add `LocoKit/Timelines` to my `Podfile` to make it build again...

sobri909, 2019-02-15:
@poulpix Sorry, I still haven't had a chance to look into this. I recommend doing what you're doing, and including the `Timelines` subspec, as a workaround for now.