sobri909/LocoKit #53: Can't I get just the improved Motion Activity?

bhulkuKs, 2019-02-18:
My project just needs near real time and accurate motion activity detection but not the location tracking with it.
It would be really nice if you could make a separate repo for it. Also for motion activity internal algorithm, if location is required then give an option to give our location manager as input parameter.

sobri909, 2019-02-19:
I'm not sure what you mean by "motion activity". Do you mean [ActivityTypeName](https://github.com/sobri909/LocoKit/blob/develop/LocoKit/Base/ActivityTypeName.swift)? If so, detecting those types is heavily dependent on location data. It is not possible to determine those types without location data. 

The only types that can be detected with any accuracy without location data are walking and running. And if you only need those types, then I recommend using Apple's own Core Motion classifier, as it is able to detect walking and running with reasonable accuracy (but not cycling or automotive with any reasonable level of accuracy).

> if location is required then give an option to give our location manager as input parameter.

LocoKit needs to carefully manage the LocationManager, adjusting factors such as desiredAccuracy, in order to achieve energy efficiency over long recording sessions. Although providing an external LocationManager could be possible, there's no obvious benefit to doing so.

If you do not need timeline data (such as Visits and Paths), then I recommend using a combination of `LocomotionManager` and `TimelineClassifier`, but without making use of `TimelineRecorder`.

You can take `LocomotionSample` instances from `LocomotionManager` at regular intervals (eg every 6-10 seconds on a timer), and determine their activity types with the `TimelineClassifier` singleton. 

```swift
let loco = LocomotionManager. highlander
loco.startRecording()

let sample = loco.locomotionSample()
let activityType = TimelineClassifier.highlander.classify(sample)
```

bhulkuKs, 2019-02-19:
@sobri909 Due to the repo structure it seems difficult to integrate the lib. I'm on xcode 10.1 swift 4.2 and one of the issue's answer suggested to point pod to develop branch. But that develop branch seems says there is no ActivityTypeName. It doesn't have some code of master branch. Also it says unable to compile objective c module LocoKit. I tried to fix as much as I can but it's yet so unstable. What mess is this? Why isn't installing a library a breeze experience? 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

sobri909, 2019-02-19:
> What mess is this? Why isn't installing a library a breeze experience?

If you would like free assistance, please approach me politely. If you prefer to be combative while asking for advice on how to use my lib that is made freely available to you, then I recommend contacting me by email first, to enquire about my hourly contracting rates.

As to repo structure, as has been recommended in other threads, you should be using the `develop` branch, and the Podfile from the Demo App shows you which pods to install.

```ruby
pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKit/Timelines', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
```

As stated above, if you require any further assistance, please contact me by email for my hourly contracting rates. Your manner is not appropriate for a situation where you are asking a stranger for assistance on how to use something that they are giving to you for free.

closed 2019-02-19 by sobri909