sobri909/LocoKit #54: Database Structure

kurry421, 2019-03-05:
Hey - really cool library. I wanted to get a better understanding on what the storage tables look like (on the sql db on the phone). Is this documented somewhere (I couldn't locate it, perhaps I'm looking in the wrong place 🤔)? 

Again, I appreciate the library you created and your response -Thank you!

kurry421, 2019-03-05:
Sorry - ended up figuring it out. `Connect device -> Window -> Device & Simulator-> Click cog icon -> download the container`.

DB is located in `AppData -> Library -> Application Support LocoKit.sqlite`. Open with your fav db software.

closed 2019-03-05 by kurry421

sobri909, 2019-03-06:
@kurry421 You can also find the database schema declared in [TimelineStore+Migrations.swift](https://github.com/sobri909/LocoKit/blob/develop/LocoKit/Timelines/TimelineStore%2BMigrations.swift).

In there you'll find all the various bits I've added to the schema in new LocoKit versions. 

I also use the LocoKit database for Arc App's storage needs, so I keep a subclass of `TimelineStore`, with Arc's own extra schema additions and customisations. 

## Example Snippets

```swift
final class ArcStore: TimelineStore, BackgroundWork {

    ...

    override func migrateDatabase() {
        super.migrateDatabase()

        ....

        migrator.registerMigration("Arc3.0.0 workouts") { db in
            try db.alter(table: "TimelineItem") { table in
                table.add(column: "workoutRouteId", .text).indexed()
            }
        }

        migrator.registerMigration("Arc3.0.0 notes") { db in
            try db.create(table: "Note") { table in
                table.column("noteId", .text).primaryKey()
                table.column("date", .datetime).notNull().indexed()
                table.column("lastSaved", .datetime).notNull().indexed()
                table.column("body", .text).notNull().indexed()
                table.column("deleted", .boolean).notNull().indexed()

                // CKObject
                table.column("ckRecordName", .text).indexed()
                table.column("ckLastSaved", .datetime).indexed()
                table.column("ckVersion", .integer).indexed()
            }
        }

        ...
    }
    ...
}
```

That lets me use a single database for LocoKit's data, and extend from that for anything custom that Arc needs.