sobri909/LocoKit #55: Xcode 10.2 Can't Build

Faydee1220, 2019-03-26:
My podfile is as below:
```
pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKit/Timelines', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
```
And I got error:
../Pods/LocoKit/LocoKit/Timelines/ActivityTypes/ActivityTypesCache.swift:10:8: Module compiled with Swift 4.2 cannot be imported by the Swift 5.0 compiler: ../Pods/LocoKitCore/LocoKitCore.framework/Modules/LocoKitCore.swiftmodule/arm.swiftmodule

../Pods/LocoKit/LocoKit/Timelines/ActivityTypes/ActivityTypesCache.swift:10:8: Module compiled with Swift 4.2 cannot be imported by the Swift 5.0 compiler: ../Pods/LocoKitCore/LocoKitCore.framework/Modules/LocoKitCore.swiftmodule/arm64.swiftmodule

My project's swift language is 4.2, and pod is also 4.2.
I already try this https://stackoverflow.com/a/46739102/5957749 but still not work at all......

sobri909, 2019-03-26:
Hi @Faydee1220! I've just installed Xcode 10.2 this morning, and will be preparing a new `LocoKitCore` build for it today. So hopefully you should be able to do a `pod update` tomorrow (or later today) that fixes the problem 😄

sobri909, 2019-03-29:
Sorry for the delay. It's all done now! 

I've tested the latest LocoKitCore in the LocoKit Demo App, and everything is working smoothly in Xcode 10.2 now 🎉

closed 2019-03-29 by sobri909

abenyahya, 2019-04-25:
Hi @sobri909, I using also the develop branch on XCODE 10.1 with Swift 4.2 and it said that the module is build using Swift 5, is there any solution for that ?

sobri909, 2019-04-26:
@abenyahya For that, you should pin to the LocoKit commit before the migration to Xcode 10.2. 

So that would be something like: 

```ruby
pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop', :commit => '7834440ebcbf9e0c7e07c345413a8453dd9d29c6'
pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop', , :commit => '7834440ebcbf9e0c7e07c345413a8453dd9d29c6'
pod 'LocoKit/Timelines', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop', :commit => '7834440ebcbf9e0c7e07c345413a8453dd9d29c6'
```

abenyahya, 2019-04-26:
So I need to migrate to xcode 10.2 ? now I'm on 10.1. I tried to pin to this specific commit, I have the same error

sobri909, 2019-04-26:
You shouldn't still have the same error. It's probably worth deleting the Pods folder and doing a fresh `pod install`.