sobri909/LocoKit #56: Project won't build when importing LocoKit

djibouti33, 2019-04-30:
Hi Matt,

First off, amazing open source contribution. Thank you so much, this is really helping my idea get off the ground.

I'm having a hard time including LocoKit into my app. I'm using Xcode 10.2.1 (10E1001). 

I was also having difficulty building the demo app, until I started browsing the Closed Issues and learned that I need to be working off the develop branch.

My pod file has this:
 ```
pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
 pod 'LocoKit/Timelines', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
 pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
```

I also ran through these steps, as posted in Issue #10:
```
On Xcode, cleaned the project and its build folder
pod cache clean --all
rm -r Pods/
pod update
pod install
```

I'm getting the following errors from the LocoKit framework:

> .../Pods.build/Production Debug-iphoneos/GRDB.swift.build/module.modulemap:11:12: Header 'GRDB-Swift.h' not found
> .../Pods/LocoKit/LocoKit/Timelines/ActivityTypes/ActivityType.swift:9:8: Could not build Objective-C module 'GRDB'
> 

Any idea what I'm doing wrong?

sobri909, 2019-05-03:
Sorry for slow reply!

I really have no idea why this would be happening. GRDB is a dependency of LocoKit, used for the SQLite database layer. The error you're seeing is not one I've seen before. The Demo App definitely builds and runs properly in Xcode 10.2.1 for me, with that Podfile configuration. 

The only thing I can think to try is to "Clean build folder", and perhaps also delete derived data in Xcode. It looks to me like Xcode is confused somehow.

keyhan76, 2019-11-15:
I had the same problem.
Fixing the warnings of the LocoKit would solve the problem.

mcauto, 2021-01-23:
I had the same problem too.
How to solve?
I get a following message in xcode.
```
ld: framework not found GRDB
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

**Podfile**
```ruby
# Uncomment the next line to define a global platform for your project
platform :ios, '14.0'

target 'test' do
  # Comment the next line if you don't want to use dynamic frameworks
  use_frameworks!
  pod 'GRDB.swift', '2.10.0'
  pod 'LocoKit'
  pod 'LocoKit/LocalStore' # optional
end
```

**Environment**

xcode: 12.3 (12C33)
platform: iOS 14.0
Swift: 5