sobri909/LocoKit #57: pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop', :commit => 'bf339cca2523d7fdb3ede1a63d8943877b469da8'   pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop', :commit => 'bf339cca2523d7fdb3ede1a63d8943877b469da8'

VadimPlasiciuc, 2019-04-30:


VadimPlasiciuc, 2019-04-30:
[image]

I'm trying to integrate locokit in swift 4.2, but didn't succeed

  pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop', :commit => 'bf339cca2523d7fdb3ede1a63d8943877b469da8'
  pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop', :commit => 'bf339cca2523d7fdb3ede1a63d8943877b469da8'

sobri909, 2019-05-03:
Hm. I've no idea why that would be happening. Although that commit you're pinning to is an odd one to pin to. But it's definitely prior to LocoKitCore being recompiled in Xcode 10.2, so it should be working fine. 

I'd recommend upgrading to Xcode 10.2 anyway. You don't have to convert your project to Swift 5.0 in order to use Xcode 10.2. You can still compile Swift 4.2 code just fine in it.

rserentill, 2019-06-11:
I'm having a similar issue, but with iOS 5:

`Module compiled with Swift 4.1 cannot be imported by the Swift 5.0.1 compiler`

When trying to import LocoKitCore

Are you planning to update the module to make it Swift  5 compatible?
Thanks!

sobri909, 2019-06-12:
Hi @rserentill. LocoKit is already Swift 5 compatible. Make sure that you're using the `develop` branch.

rserentill, 2019-06-12:
Oh.. I was using it with CocoaPods. I guess it is using the master branch. I'll try it using develop. Thanks!

caohrong, 2019-06-25:
Why not compatible in master branch? So How should  I support to do if I want use it in CocoaPods?? Thanks.

sobri909, 2019-06-25:
> Why not compatible in master branch?

Because I do not have enough spare time or funding to focus on doing another LocoKit stable release yet. 

If you would like to see more progress on LocoKit development, then please consider becoming a backer for the project [on Patreon](https://www.patreon.com/sobri909).

> How should I support to do if I want use it in CocoaPods?? 

Taken from the `Podfile` of the LocoKit Demo App in the `develop` branch: 

```ruby
pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKit/Timelines', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
```

sobri909, 2019-06-25:
Please note that the develop branch is also considered stable. All Arc App releases on the App Store use the develop branch of LocoKit, so that branch is already in daily use by tens of thousands of end users. 

The reason why I have not done another merge to master, and published another release to CocoaPods, is because I do not have time to update the documentation. The develop branch has many additional bits of API that require documentation.

caohrong, 2019-06-25:
Thanks you so much.😊

StasMalinovsky, 2019-09-18:
Hello! I'm getting the same error on `import LocoKitCore`line in **Xcode 11.0 beta 6**. 

_Module compiled with Swift 5.0.1 cannot be imported by the Swift 5.1 compiler:_ _/Users/stas.malinovsky/Projects/LocoKit/Pods/LocoKitCore/LocoKitCore.framework/Modules/LocoKitCore.swiftmodule/arm64.swiftmodule_

Any suggestions?

sobri909, 2019-09-19:
Looks like I need to build another LocoKitCore binary! 

Also, @StasMalinovsky, you should probably download the Xcode 11 GM seed 2. Although presumably the same new LocoKitCore binary will work for both. 

I'll try to get that done today.

closed 2020-01-16 by sobri909