sobri909/LocoKit #61: Question about deep sleeping

StasMalinovsky, 2019-07-02:
Hello! Just wanted to say thanks for this awesome lib!
There is a question. I see that in developer branch in `LocomotionManager` there is a **public func** `startDeepSleeping(until wakeupTime: Date)`, but there is no function calls in all project. Where and when this function should be called? Thanks!

sobri909, 2019-07-04:
> Hello! Just wanted to say thanks for this awesome lib!

Thanks 😄🙏🏼

Sorry for slow reply on this one. It slipped through my inbox. 

> I see that in developer branch in LocomotionManager there is a public func startDeepSleeping(until wakeupTime: Date), but there is no function calls in all project. 

Deep sleep is a bit of a dangerous feature, especially for third party apps. It allows the app to be suspended completely by iOS. Which means that there is high risk that recording won't restart at the desired time. 

The way that deep sleep works, to get the app out of suspended state / relaunched, is by use of background refresh, and silent push. The silent push part isn't available to third party apps yet, because it requires registering the app with LocoKit's server. Currently the server only knows how to send pushes to Arc App.

The background refresh part should work fine in third party apps. But I wouldn't personally want to rely on that alone, to get an app relaunched / out of suspended state. Although it'd be fun to experiment with, and see how it goes.

> Where and when this function should be called?

Arc App goes into deep sleep state when its place ML models say that the user is very unlikely to go anywhere for the next 2+ hours. It then enters deep sleep mode, with the wakeupTime set to the ML model's predicted leaving time (minus a few minutes, to get things started before they head out).

StasMalinovsky, 2019-07-18:
Oh, sorry too for slow reply :)
Thanks for this answer to my question.

I found out that the background app refresh does not work at night(bedtime is turn on on iOS device ), but looking through similar applications(for example life 360), I see in the battery usage that the application has background activity for one minute once in hour (the device at this time is stationary, I sleep :))
May be you ever face issue like that? What the way to wakeup the app at night?

sobri909, 2019-07-19:
Whether LocoKit shows as active for 1 minute or 60 minutes in the background overnight will make no difference to battery consumption. LocoKit's sleep mode recording has almost zero battery impact.

The only thing that can achieve greater battery life overnight is allowing the phone to go into Standby Mode. For the phone to go into Standby Mode it requires that all background apps are suspended. 

If Life360 is not suspended, then the phone can't go into Standby Mode. If LocoKit is not suspended, then the phone can't go into Standby Mode. 

LocoKit's Deep Sleep Mode is designed to allow the app to be suspended. That then allows the phone to go into Standby Mode (provided no other apps are still active in the background). That will then achieve excellent battery savings overnight.

sobri909, 2019-07-19:
LocoKit can record in Sleep Mode for 24-48 hours on older / smaller model phones, and 48-72 hours on newer / larger model phones, on a single battery charge. That is about the same amount of time the phone can survive with no apps running at all, but with Standby Mode disabled. 

So basically what that means is that LocoKit's sleep mode has almost zero battery cost, with the only cost being that it stops the phone from going into Standby Mode. If you want greater battery savings overnight, Standby Mode is what you want to achieve. If the app is active for one minute every hour, the phone will not be able to go into Standby Mode, so that kind of algorithm doesn't offer greater battery savings.

closed 2020-01-16 by sobri909