sobri909/LocoKit #63: Can't build develop

corradio, 2019-07-30:
Hello,

I've just added the develop branch as pod dependency using
```
pod 'LocoKit', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
  pod 'LocoKit/Timelines', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
  pod 'LocoKitCore', :git => 'https://github.com/sobri909/LocoKit.git', :branch => 'develop'
```

When building, I get this error in LocoMotionManager.swift:
[image]

It seems to be due to the fact that LocoKitCore framework hasn't been committed to the repo since a long time?
Is it possible to build it from source?

sobri909, 2019-07-31:
Ugh. My bad. I must've pushed some LocoKit commits to Github that weren't supposed to be pushed until I'd built a new LocoKitCore 😣 Sorry about that. 

I'll get an updated LocoKitCore built and live today.

sobri909, 2019-08-01:
Ok, there's an updated LocoKitCore now, so develop branch builds and runs again. I've tested it in the Demo App, and all is going smooth. 

Again, sorry about that! I must've pushed some stuff live too early, thinking I was in another repo.

closed 2019-08-01 by sobri909

corradio, 2019-08-01:
Terrific thank you! Is there a reason why the LocoKitCore is not open source?

sobri909, 2019-08-01:
Yeah, a couple of reasons. 

The first is that it manages the secure communications with the LocoKit server for fetching activity type ML model data. 

The second is that it contains ActivityBrain, which does the low level filtering and processing of raw data (location, accelerometer, pedometer, etc). The algorithms in there are the result of a lot of R&D, and I'm not aware of any competing SDKs that are capable of LocoKit's levels of raw data filtering, smoothing, and real time stationary/moving state detection. So I'm not super keen to publish those algorithms publicly yet. 

I will probably move more of LocoKitCore out into the public code in LocoKit over time though. For example the dynamic sample smoothing could go open source. Although that's quite tightly coupled with the real time moving/stationary state detection, so open sourcing one without the other would be quite a challenge.

corradio, 2019-08-01:
That makes sense. I've worked on similar topics for a couple of years and would be keen to contribute but I see the trade-off you have to make.
In any case thank you for making this part open source. I'd love to see if I can help with specific issues.

Olivier

sobri909, 2019-08-01:
Thanks 😄

Yeah, I'd really love to make the whole thing open source. I'm just wary of competing commercial SDKs (or Arc App competitors) jumping on the algorithms and copying them into their closed source products then claiming them as their own. I don't have the money to go after anyone legally, so I'd be pretty powerless other than to say "Please don't steal my hard work".

That's also why LocoKit is licensed as LGPL rather than MIT/BSD. Competitors are welcome to use it, but if they do use it, and modify it to their own purposes, legally they have to make those changes public.