sobri909/LocoKit #64: Xcode 11, Swift 5.1

tomdaniel0x01, 2019-09-22:
Xcode 11 gives error 'Module compiled with Swift 5.0 cannot be imported by the Swift 5.1 compiler'

sobri909, 2019-09-26:
Apologies for the delay on this. I was trying to get LocoKitCore migrated to the new XCFrameworks, but failed miserably. (It needs to be supported by CocoaPods, because the LocoKit pod depends on the LocoIKitCore pod, and [CocoaPods hasn't got that far yet](https://github.com/CocoaPods/CocoaPods/issues/9148).)

I've gone back to the old hackish way of doing it for now, and pushed that live instead. 

Just testing it now in the Demo App, and ... yep, it's working. Phew. 

Not as cool as XCFrameworks, and that *will* need to be done eventually. But it looks like I jumped the gun by trying to get it done now. 

LocoKit / LocoKitCore should also migrate to (or add support for) Swift packages soon too. Yet another housekeeping todo that doesn't involve actually building new features. Sigh 😏

sobri909, 2019-09-26:
Ok, yep, confirmed that it's all good and working. Closing this one for now. Will have to deal with the XCFramework stuff later, once CocoaPods is ready. (Or perhaps just migrate completely over to Swift packages, if that's possible / sensible).

closed 2019-09-26 by sobri909