sobri909/LocoKit #65: The necessary unit tests should be created

Li-Qun, 2019-11-10:
Why don't you create some testUnit of the classes, I think that we'll understand how they work better and faster.

sobri909, 2019-11-10:
LocoKit doesn't have regression issues, so automated unit testing would not provide any benefits to the project. 

For understanding the API, please let me know which parts of the API you are interested in and I will focus on filling in the documentation for those areas. Thanks 😄

Li-Qun, 2019-11-14:
I am interested in that 
（1）How do you determine which points are a group？（rawLocations => each LocomotionSample process）
 (2)   Determine the point  moving/stationary of  specific process
 (3)  you have you own movingType and how do you use coreMotionType

Thanks  a lot . Waiting for your reply

sobri909, 2019-11-18:
>（1）How do you determine which points are a group？（rawLocations => each LocomotionSample process）

This is done inside the closed source LocoKitCore framework, and is a proprietary system 😉 However I can give you a simple outline of how it works: 

The required number of raw CLLocation objects necessary for each LocomotionSample is dynamically adjusted baed on current horizontalAccuracy (ie better accuracy requires less raw locations). There are also other factors that go into deciding the duration of each LocomotionSample. 

You can learn more about LocomotionSamples from [the API documentation](https://www.bigpaua.com/locokit/docs/Classes/LocomotionSample.html).

> (2) Determine the point moving/stationary of specific process

Again, this is done inside the close source LocoKitCore framework, and is proprietary. 

> (3) you have you own movingType 

LocoKit's activity types are determined by the ML classifiers. I recommend looking at the sample code in the LocoKit Demo App to get an understanding how these work.

> and how do you use coreMotionType

LocoKit doesn't use coreMotionType at all. It is far too inaccurate, and is worse than useless. Making us of it in the ML, or for any other purpose, worsens the accuracy of the results. It is included in the LocomotionSamples purely for informational and comparison purposes.

Li-Qun, 2019-11-26:
Bravo!!!  Chinese called:厉害了!!!  I still want to know that how to determine the location status is moving or stationary,  that data is inaccurate, just speed between [0 - 3km/h)  and the radius is [10 150m]  two elements right?? or how do I determine one location status? then I think i could find some stationary poinst and show them.Because I need to use my own database, to control the lcoation data freedom,but locokit dosen't supply the fundation, i need write it by myself.

sobri909, 2019-11-26:
The moving / stationary state detection is as accurate as possible given the available raw data. You are unlikely to be able to achieve higher accuracy yourself, although I’d love to see someone try!

Speed isn’t a factor in the algorithm, except as a product. There are a bunch of dynamically determined factors used, adjusted in real time to suit available location data accuracy. The higher the current location data accuracy, the more real time the stationary / moving state detection, and the greater the granularity. Slower movement combined with worse available raw data will result in less granularity and less real time. 

In the LocoKit Demo App pay attention to the current sample’s “behind now”, which tells you how far into the past the moving state refers to. With higher quality raw data, the “behind now” will be shorter, thus representing the user’s state closer to real time. 

I recommend you also make use of TimelineRecorder and TimelineItems (Paths and Visits). Those represent the highest possible accuracy available at any point in time. 

Building a system parallel to LocoKit will almost certainly be wasted time unless you have many spare months to put into R&D, or already have substantial background in this area. Although perhaps it will help you to try building such a parallel system, in order to understand the complexities and the mathematical limitations given the available raw data 😉

closed 2020-01-16 by sobri909