sobri909/LocoKit #73: Is TimelineManager still available?

annawidera, 2020-10-06:
Hello Matt (@sobri909), thank you for the great open-source contribution!

I've started trying out the `LocoKit` however experienced some troubles while following the readme. 
Is `TimelineManager` still available in `LocoKit`? Although it is mentioned in the README, I had no luck adding it's support to the `LocoKit Demo App`.

> Cannot find type `TimelineManager` in scope
[image]

```swift
import LocoKit
import Anchorage
import SwiftNotes
import CoreLocation

class ViewController: UIViewController {

    let store = TimelineStore()
    
    var recorder: TimelineRecorder
    var dataSet: TimelineSegment?
    
    var timelineManager: TimelineManager

(...)

    // MARK: controller lifecycle

    init() {
        // LocoKit's Activity Type Classifiers require an API key
        // API keys can be created at: https://www.bigpaua.com/locokit/account
        // LocoKitService.apiKey = "<insert your API key here>"

        if LocoKitService.apiKey != nil {
            ActivityTypesCache.highlander.store = store
            recorder = TimelineRecorder(store: store, classifier: TimelineClassifier.highlander)

        } else {
            recorder = TimelineRecorder(store: store)
        }
        
        self.timelineManager = TimelineManager()
        
        super.init(nibName: nil, bundle: nil)
    }
```

I've tried also to use `PersistentTimelineManager` but it wasn't found either. (For that I've installed `LocalStore` by adding `pod 'LocoKit/LocalStore'` to the `Podfile`)

Any help will be greatly appreciated! 😊 
---
`PersistentTimelineManager` is mentioned [here in the changelog](https://github.com/sobri909/LocoKit/blob/master/CHANGELOG.md#added-1).

`TimelineManager` is mentioned [here in the README](https://github.com/sobri909/LocoKit#record-timelineitems-paths-and-visits) and [here in the docs](https://www.bigpaua.com/locokit/docs/Classes/TimelineManager.html)

sobri909, 2020-10-07:
Hi @annawidera! Unfortunately the docs are woefully out of date 😞

The best way forward these days is to pull a local copy of Arc Mini (https://github.com/sobri909/ArcMini) and use that as example code. 

Arc Mini is a fully functioning app that makes use of almost all of LocoKit's functionality. It's also got a bunch of classes that I'm planning to move upstream to LocoKit (those tasks are tracked [here](https://github.com/sobri909/ArcMini/projects/1?card_filter_query=locokit)). 

I'd also possibly recommend using LocoKit's `develop` branch, depending on your needs. I haven't had time to do another LocoKit stable release in a few months, so develop branch is ahead in some specific areas. Though hopefully they haven't diverged far in terms of API yet...

sobri909, 2020-10-07:
This is the full setup I'm using in Arc Mini's AppDelegate at the moment: 

```swift
LocoKitService.apiKey = "<api key>"
ActivityTypesCache.highlander.store = RecordingManager.store
LocomotionManager.highlander.coordinateAssessor = CoordinateTrustManager(store: RecordingManager.store)
LocomotionManager.highlander.appGroup = AppGroup(appName: .arcMini, suiteName: "group.ArcApp")
LocomotionManager.highlander.appGroup?.timelineRecorder = RecordingManager.highlander.recorder
```

Though all of those lines are optional. The first two lines are only if you're using LocoKit server API provided activity type models for sample activity type classification.

The third line is only useful if you're allowing users to manually classify samples / timeline items (UI to allow them to correct their recorded timelines). The `coordinateAssessor` provides "Trust Factor" values for coordinates, based on samples that have been classified as "stationary" at/near that coordinate. Basically "stationary" samples that have non-zero speeds indicate that location data is less trustworthy at that coordinate. The higher the speed of stationary samples, the less trustworthy the location data at/near that coordinate. 

The last two lines are only useful if you're running multiple LocoKit based apps in parallel that are sharing their databases. So that's really niche and almost certainly not necessary.

sobri909, 2020-10-07:
In Arc Mini's code, have a look at the `RecordingManager` class. That's my high level manager class for wrapping up the data store, timeline recorder, and various related activities.

sobri909, 2020-10-07:
Oh and a little history on why `PersistentTimelineManager` isn't around anymore: The old non persistent timeline manager class went away, after it became too difficult to juggle and keep up to date both persistent and non persistent stores. So now all stores are persistent, SQLite backed, and `PersistentTimelineManager` got renamed to `TimelineManager` after the old one went away.

sobri909, 2020-10-07:
Oh, I should mention `ArcStore` too. That's Arc Mini's subclass of `TimelineStore`, which is something that's necessary to do if you want to subclass TimelineObjects. In Mini's case that's `ArcVisit` / `ArcPath` as subclasses of `Visit` and `Path`. 

ArcStore also provides the Place models that aren't part of LocoKit yet. Plus some other stuff. There's a lot going on in there. It's perhaps not a great idea to jump into a TimelineStore subclass initially 😏 Best to save that for when it's needed.

closed 2020-12-07 by sobri909