sobri909/LocoKit #74: LocoKit won't build on Apple Silicon Macs

bogo, 2020-12-03:
Thank you for an awesome library, @sobri909!

I recently moved to using an M1 Mac as my daily driver - and, unfortunately, `LocoKitCore.framework` is not a happy camper, due to a missing ARM64 version of the iOS 10 SDK and Swift 5.1/5.3 incompatibilities. Attempts to build an app with LocoKit end up with the following error:

```<unknown>:0: error: unable to load standard library for target 'arm64-apple-ios10.0'
[redacted]:10:8: error: failed to build module 'LocoKitCore' from its module interface; the compiler that produced it, 'Apple Swift version 5.1.3 effective-4.2 (swiftlang-1100.0.282.1 clang-1100.0.33.15)', may have used features that aren't supported by this compiler, 'Apple Swift version 5.3.1 effective-4.2 (swiftlang-1200.0.41 clang-1200.0.32.8)'
import LocoKitCore
```

Any chance you could rebuild the `LocoKitCore.framework` with the most recent Swift compiler and/or target a newer standard library? Happy to help out with debugging this if you don't have access to an Apple Silicon Mac!

sobri909, 2020-12-04:
Ah that makes sense. What I'll do is prioritise the open sourcing of LocoKitCore. I've been meaning to get it done for a year or more now, but just haven't found the time. This will give me justification to spend time on it.

sobri909, 2020-12-04:
Ok I've got LocoKitCore open sourced now (by just moving the entire contents of it into LocoKit proper). That means that LocoKitCore is no longer a required dependency. But I'll need to update the podspec etc. And this also means I can easily make LocoKit a SwiftPM package now, so I'll do that too (as long as it goes smoothly).

sobri909, 2020-12-04:
OK I've finally got the SwiftPM Package.swift working properly (I think). Uh... I probably shouldn't tap the send button on this reply until I've broken that everything still compiles and runs, in a project using LocoKit via SwiftPM instead of CocoaPods. Let's see how this build and compile goes...

Uh, well, that didn't go great, but it's Arc repo specific so far. I'm doing a lot of `import LocoKitCore` in Arc App's code, because I use some of LocoKitCore's (now LocoKit proper's) helper extensions and methods. I'm gonna have to go through and remove all those import statements. 

But otherwise it *seems* solid. I'll clean up more of this tomorrow. For now, you should be able to add LocoKit as a SwiftPM package, with no binary dependencies, and build on the M1. Make sure you remove LocoKit from your Podfile too, to avoid super weirdness.

sobri909, 2020-12-04:
Oh wait, it looks like LocoKit itself has some `import LocoKitCore`. Uh, I'm running out of time today so I'm not going to get a chance to do a commit with those calls removed tonight. But if you're in a hurry, you should hopefully be able to just rip those imports out and all will be well. *Hopefully*.

sobri909, 2020-12-05:
Got it! It took a whole bunch of `import Foundation`, and explicit platform specification in the Package.swift, and some other fiddling, but it's all there now. 

LocoKit is finally all source, all open source, and usable as a SwiftPM package. Woo!

closed 2020-12-07 by sobri909

sobri909, 2020-12-07:
Oh, I should mention a couple of changes to packaging: 

1) I'm no longer updating the CocoaPods package(s). The Upsurge dependency made it impossible to keep them up to date a while back anyway, due to that project only keeping their SwiftPM package up to date, and their last pod release no longer passing `pod trunk push` validation due to Swift version stuff. 

2) For the SwiftPM package, you will want to pin to a branch, not a tag/release, because I'm simply too overworked and burnt out to manage doing proper releases at the moment. I recommend pinning to `master` unless you're building on top of Arc Mini (in which case pin to `develop`). Arc Mini's development happens closely in parallel with Arc App (both v3 and v4), with those project occasionally requiring LocoKit changes, which get pushed directly to the develop branch.