sobri909/LocoKit #80: ClassifierResults only returns one value: Car

JohannesSteimel7P, 2021-09-02:
Hi Matt,
I'm currently trying out your LocoKit Demo App. Yesterday I generated an API key and startet tracking but the results in the ClassifierView contained only one ClassifierResultItem (car) instead of multiple.

**ClassifierResults(results: [LocoKit.ClassifierResultItem(name: LocoKit.ActivityTypeName.car, score: 0.0, modelAccuracyScore: Optional(0.965837))], $__lazy_storage_$_arrayIterator: nil, moreComing: false)**

At first I thought I did something wrong but then a took a small walk with the Arc App where it even counted my steps but still categorized it as "car". Am I missing something or is there currently an issue with the LocoKit API?

[image]

sobri909, 2021-09-02:
Hmm. It's a long time since I've used the Demo App, so I'm not sure if it's using up to date code. I recommend trying the Arc Mini project instead, as a more comprehensive demo app, with fully up to date code. 

https://github.com/sobri909/ArcMini

See how you get on with that one.

JohannesSteimel7P, 2021-09-07:
Thanks for the quick response. For now I'm going to use Apples CMMotionActivityManager but maybe I'll have a look at ArcMini later on.

closed 2021-09-07 by JohannesSteimel7P