sobri909/LocoKit #81: Getting Only stationary from mostRecentItem's activityType

avinashamanjha251, 2021-09-30:
Hi Matt, 
I'm trying to fetch most recent "TimelineItem" and I'm getting it... but every time I'm getting activityType either getting stationary or nil. Please help me on this. Thanks in advance. 

 if let mostRecentItem = self.store.mostRecentItem {
            printDebug("=========================================")
            printDebug("item \(mostRecentItem)")
            printDebug("item modeMovingActivityType \(String(describing: mostRecentItem.modeMovingActivityType))")
            printDebug("item activityType \(String(describing: mostRecentItem.activityType))")
            printDebug("item modeActivityType \(String(describing: mostRecentItem.modeActivityType))")
            printDebug("item startDate \(String(describing: mostRecentItem.startDate))")
            printDebug("item endDate \(String(describing: mostRecentItem.endDate))")
            printDebug("item duration \(mostRecentItem.duration)")
            printDebug("item stepCount \(String(describing: mostRecentItem.stepCount))")
}

sobri909, 2021-09-30:
Hi @avinashamanjha251!

The first thing I'd check is whether samples are being classified at recording time. The `item.activityType` getter relies on either cached `classifierResults` or on the `classifiedType` of each sample within the item. So if you haven't explicitly touched `item.classifierResults` to trigger full classifier results for the item, then `item.activityType` won't do so either, and will fall back to `modeMovingActivityType`/ `modeActivityType`. Those two rely on the individual samples already having been classified, typically at time of recording.

As long as the `TimelineRecorder` was initialised with a classifier, eg `let recorder = TimelineRecorder(store: store, classifier: UserTimelineClassifier.highlander)`, and the device has an internet connection and valid LocoKit API key, the recorder will be classifying each sample at recording time. 

Though `item.classifierResults` also relies on the recorder having a valid classifier assigned, so that would be the second thing to check: whether you get sensible results by going direct to `item.classifierResults`. 

It'd also be worth printing the `sample.classifiedType` of each sample within the item. eg `for sample in path.samples { print("sample.classifiedType: \(sample.classifiedType)") }`. The results of that should give better insight into what's going on with `modeMovingActivityType` / `modeActivityType`.

Let me know you get on!

avinashamanjha251, 2021-09-30:
@sobri909 .. actually I'm not using classifier 

 lazy var recorder :TimelineRecorder = {
        let recorder = TimelineRecorder(store: store)
        return recorder
    } ()

sobri909, 2021-09-30:
Ah, without a classifier, samples and items won't have a activity types. That'll be the problem 😉