sobri909/LocoKit #83: Xcode 13 / Swift 5.5 compatible LocoKitCore

mwoollard, 2021-11-03:
Current CocoaPods published version is 7.0 (although I see 7.1 tagged here in github).

With Xcode 13 / Swift 5.5 am getting:

`error: module compiled with Swift 5.1.3 cannot be imported by the Swift 5.5.1 compiler`

Will you be providing a compatible update for the binary LocoKitCore ?

Thanks

sobri909, 2021-11-04:
I recommend switching to using Swift Package Manager. I'm treating CocoaPods as legacy, at this stage, and not testing to make sure it still works. 

> Will you be providing a compatible update for the binary LocoKitCore ?

Ah, are you still on the `master` branch? I also recommend switching to the `develop` branch. Master hasn't been updated in many months, but the develop branch is fully stable and ready to go.

sobri909, 2021-11-04:
I might work to do another stable release some time before the end of the year. I'm working on migrating some parts of Arc Mini upstream to LocoKit (UserActivityTypeClassifer etc, and some logging stuff). But yeah, for now I recommend switching to SwiftPM and using the develop branch.

mwoollard, 2021-11-04:
OK thanks, will give it a go.

sobri909, 2021-11-04:
Of note: In the develop branch there's no more binary requirement. LocoKitCore is no longer closed source. So that makes the packaging shuffle a lot simpler. (I thought this was also live in the master branch too! So that branch must be even more out of date than I thought 😩)