sobri909/LocoKit #86: Latest 3.9.1 with iOS 16.4.1 extreme battery drain

ss32nji3, 2023-04-13:
I’ve noticed that my battery was draining for no reason. Went to battery and saw arc consumed 20% of the battery in the past 24 hours. iPhone 14 Pro, iOS 16.4.2. 

[image]

sobri909, 2023-04-14:
It's possible that [this commit](https://github.com/sobri909/LocoKit/commit/ec1dcce78ea900909b5def6610ea93a20548aac9) is the culprit. Though iOS Setting app's Battery view is notoriously unreliable on a day to day basis, so it'll be worth keeping track of a week or so of daily results before drawing any conclusions. 

Would also ideally want to compare to efficiency with pre-iOS 16.4, which obviously won't be easy to do. But yeah, if it looks like there's an ongoing problem, it'll be necessary to escalate it to Apple as an iOS bug. The change they made in iOS 16.4 forced the subsequent change in LocoKit, which definitely has a risk of using more energy each day.

sobri909, 2023-04-14:
Oh, also important to keep in mind that the Battery view is showing relative percents, not absolutes. So that 20% is relative to the other apps / services in the list.

If "Home & Lock Screen" is second in the list, it means your phone was likely idle for most of the past 24 hours (ie it likely spent more time showing Home Screen than any actual apps). So Arc using 20% might actually not be a big deal in that context.

Probably the easiest way to quantify whether things are actually genuinely worse or not will be to keep track of when you charge the phone and how many hours it takes from full charge to full discharge. That will be a more useful metric than the relative percents in the Battery view.

There's also an app, [Battery Grapher](https://apps.apple.com/us/app/battery-grapher/id1498632956?platform=iphone) which will intelligently calculate for you your average battery life per day. It will take a few days for it to collect enough data.

ss32nji3, 2023-04-14:
Ever since I started using it a few months ago the maximum battery it would use in the same battery section would be MAX 2%-3% total battery usage. I don’t have a screen shot but I remembered that it barely used any battery which I found surprising for the way the app functions.   
  

  
I know Apple has to fix it but I just wanted you to know it has been using barely any battery previously.   
  

  
  
 Thanks,  

  
Jay

sobri909, 2023-04-15:
The problem is that the Battery view's percent is only relative to other apps within that day, and can't be compared between days. For example you could have an app show 2% one day then 20% the next day, but in reality it used less battery on the 20% day than on the 2% day. It's a very confusing view!

So yeah, your best bet is to use Battery Grapher, and mostly ignore the Battery view in iOS Settings app. (Unless you want to do complex maths each day, to try to convert the relative percents to absolute percents, so that you can compare them between days. Which would be very not fun).